OpenIPSL/OpenIPSL #14: WT3 example model won't check

tinrabuzin, 2016-04-19:
Due to some parameters which are not defined, the example won't check.

MaximeBaudette, 2016-05-11:
Shall we then rename this issue ? to PSSE WT Fix ?

tinrabuzin, 2016-05-11:
Yes, we can do that.

lvanfretti, 2020-05-21:
This is being taken care of by Marcelo and Manuel, will be resolved in v2.

closed 2020-05-21 by lvanfretti

marcelofcastro, 2020-05-21:
@lvanfretti, let's wait for the pull requests to be merged. After that we can check if that problem was or was not solved. Isn't it better to wait?

lvanfretti, 2020-05-21:
@marcelofcastro you can mention this issue when the day comes.
A lot of these issues were opened by people that are no longer actively working on OpenIPSL, unless there is someone to take care of the problem, I'm closing the issue. There is not point to have a bunch of open issues.