OpenIPSL/OpenIPSL #16: Publish OpenIPSL also on impact (and therefore modelica.org/libraries)

dietmarw, 2016-04-20:
Just a small reminder that as soon as you feel ready that I can add OpenIPSL also to the impact package index on impact.github.io

MaximeBaudette, 2016-04-20:
I was discussing with @tinrabuzin and we feel that we are very close to cleaning the library from all "model check" errors. When we have that done, we will make a release of it and you can add it on impact.

tinrabuzin, 2016-08-19:
@MaximeBaudette @dietmarw I think the library can now be included on impact?

MaximeBaudette, 2016-08-19:
Yes ! I'll prepare it!

MaximeBaudette, 2016-08-19:
@tinrabuzin now I remember why I haven't done it so far...
I needed to implement the updated copyright notice. Now that OM have corrected the bug named in #47 I will be able to take care of that! 
Once the copyright is updated, I will also save the library once again in Dymola to fix the formatting and update the version number, and prepare a new release. 
Then @dietmarw can add it to Impact!

MaximeBaudette, 2016-11-16:
@dietmarw 
Now we are ready to include OpenIPSL to Impact. I will create the version v1.0.0 before the end of the week.

MaximeBaudette, 2017-02-21:
Ping @dietmarw 
I tried searching on impact.github.io but didn't find the library. Any thing we still need to fix?

dietmarw, 2017-02-21:
Well actually creating a v1.0.0 release perhaps? ;-)

MaximeBaudette, 2017-02-21:
I thought I did it:
https://github.com/SmarTS-Lab/OpenIPSL/releases/tag/v1.0.0
What did I do wrong then :cry: ?

dietmarw, 2017-02-21:
My bad, I overlooked that one since the title sad nothing wit a version number. There is a different issue now coming up which currently is preventing OpenIPSL from not showing up in the impact index. Nothing you can/have to worry about. I'm on it.

MaximeBaudette, 2017-02-21:
Great News! 
Please let me know if there is more I need to do or if I can close the issue.

dietmarw, 2017-02-21:
Leave it open for now so I can close it once I fixed the impact issue.

dietmarw, 2019-01-22:
OpenIPSL is listed on modelica.org/libraries. For technical reasons it will not appear on the impact index. The impact index might undergo a substantial redesign which will most likely also fix the listing of OpenIPSL. So closing this isssue now.

closed 2019-01-22 by dietmarw