OpenIPSL/OpenIPSL #26: Automatic Validation with Travis CI

MaximeBaudette, 2016-05-03:
Issue to discuss creating a framework for adding the possibility of automatic validation of the model by the CI server.

MaximeBaudette, 2016-05-09:
Some of the components are failing cause of a problem in OM with replaceable components:
https://trac.openmodelica.org/OpenModelica/ticket/2079

MaximeBaudette, 2016-05-11:
Current implementation of the CI does NOT produce an error when the library fails to load

tinrabuzin, 2016-08-19:
@MaximeBaudette I think we fixed this problem with replaceable components? I don't remember how anymore...

lvanfretti, 2020-05-21:
Being replaced with new CI by Marcelo and Manuel in the next release.

closed 2020-05-21 by lvanfretti

marcelofcastro, 2020-05-21:
> Being replaced with new CI by Marcelo and Manuel in the next release.

@lvanfretti I think the CI will be replaced in v2.0 not v1.9...

lvanfretti, 2020-05-21:
@marcelofcastro fine, but still, it won't be related to this issue.
Nobody is taking care of it, and it will be replaced, so I'm closing this.