OpenIPSL/OpenIPSL #30: Expandable bus causing problems in OM

tinrabuzin, 2016-05-10:
I noticed that in OM for examples N44 and SevenBus, expandable bus causes the number of equations to be higher than the number of variables in the model and, thus, the systems cannot be simulated.

tinrabuzin, 2016-05-10:
I replaced all of the expandable buses with the regular ones and the seven bus simulated with the results that are the same as the ones in Dymola.

MaximeBaudette, 2016-05-10:
I don't know how to proceed about that... 
Because on one side, the expandable bus makes it very nice to have a good overlook of the system. 
On the other side, it prevents simulation in OM... Should we just duplicate the models and use simple buses in the duplicate model?

tinrabuzin, 2016-05-10:
@MaximeBaudette Expendable bus works in the example from MSL so I guess this is due to our implementation. I'll investigate it a bit more.

MaximeBaudette, 2016-05-10:
ok

closed 2016-05-11 by MaximeBaudette

tinrabuzin, 2016-05-11:
Both N44 and SevenBus can now be simulated in OM after all of the fixes.

PR #33 (2016-05-11): Updating BusExt
Derived from Jan's PR #31 to separate the different issues reflected in that PR. Also, fixed a small bug in the connect statements and removed no longer used PwPinExt. close #30