OpenIPSL/OpenIPSL #53: Request for a Release for the 2016-05-25 ENTSO-E IOP Webinar on OpenIPSL

lvanfretti, 2016-05-24:
Hi @MaximeBaudette and @tinrabuzin 

Today I gave a webinar to the ENTSO-E IOP group regarding OpenIPSL.
I was wondering if you can make a release for them with my slides, the Tutorial package, and other materials.

I was wondering if we can create a top-level folder with the Tutorial.mo package so that it gets included in the CI tests and that is always working with the latest version of the library.

I will also ask Bernhard and Francesco to send me their slides, so we can include them as part of the Webinar.

Please send me an email when this is ready to send it to ENTSO-E.

Best

Luigi

MaximeBaudette, 2016-05-24:
I don't really understand what you want in *Tutorial.mo*:
 1. Some reference examples used in the tutorial ?
 2. A striped-down version of the library + examples compiled in a single file 

In case 1, that can be done by creating one additional */Application Example* that we could call tutorial. 
In case 2, I'm opposed to it.
The tutorial should use the library as it is, instead of requiring to develop and maintain a separate package. This would only create more work for us, instead of simplifying things. 

A possible compromise?
We could add a *.mos* script that takes care of loading the necessary files in dymola for the tutorial.

lvanfretti, 2016-05-24:
Now it is not necessary to used a stripped library.
It is just the examples.

Can you check if it would work in this situation?

MaximeBaudette, 2016-05-24:
I'll look at it later in the afternoon.

PR #54 (2016-05-25): Tuto update
Adding files for creating a release dedicated to a new tutorial

Closes #53
closed 2016-05-25 by tinrabuzin