OpenIPSL/OpenIPSL #55: Incompatibility Jmodelica and PwLine model ?

MaximeBaudette, 2016-05-26:
Today I was trying to generate some FMUs using Jmodelica and I got the following error message from Jmodelica for the PwLine model: 
> The right and left expression types of equation are not compatible, type of left-hand side is Complex, and type of right-hand side is Integer. 

About the line: 
> is = 0 + j*0;

It seems that the a+*j*b syntax is not well supported by Jmodelica, despite presenting no issues in OM or Dymola. 
Anyone has some experience on the question ?

janlav, 2016-05-26:
Maybe it should be noted that `is = Complex(re=0,im=0)` works.

MaximeBaudette, 2016-05-26:
Yes true, 
Though I find the `j`syntax to be easier to read, and also it is the documented way of using the complex package see [OM Link](https://www.modelica.org/news_items/modelica-standard-library-3.2-released). 

So I'm not sure we should change the syntax if it actually is JModelica that has a problem.

MaximeBaudette, 2016-05-26:
Seems like @fran_jo got the same issue but reported it to JModelica:
http://www.jmodelica.org/27707

MaximeBaudette, 2016-05-26:
Update:

It seems Jmodelica has problems dealing with complex numbers when the imaginary part is 0, using a syntax different from Complex(re,im).

lvanfretti, 2016-05-26:
@MaximeBaudette SystemModeler is also not working with the complex number line model.

I think we need to support the older line model, and have the complex number one as an option. 
When the Modelica compilers (other than Dymola and OM) don't have a problem with this, it should be possible to switch.

Maybe the line model itself could have a flag that detects which IDE is being used and enables the equations that are supported/working.

lvanfretti, 2020-05-21:
JModelica is no longer developed by Modelon and we will not deal with it going further.

closed 2020-05-21 by lvanfretti