OpenIPSL/OpenIPSL #58: Wrong link in readme.md

sorrento, 2016-05-31:
to the wiki
in the section "notice to users"
https://github.com/SmarTS-Lab/OpenIPSL/blob/master/update%20link

MaximeBaudette, 2016-06-02:
I'm working on updating the documentation, but it has been slower than expected, sorry for the inconvenience

lvanfretti, 2016-06-02:
I hope to have time in July also to contribute...

sorrento, 2016-06-02:
Sorry I didn't explain well:
In the readme.md, on the section  https://github.com/SmarTS-Lab/OpenIPSL#notice-to-users
the link to the wiki there leads to a "**error 404 page not found**".
I just say, since you already have some content on the wiki, put the correct url, that is 
https://github.com/SmarTS-Lab/OpenIPSL/wiki
 That's all.

Thanks!

PR #72 (2016-08-25): Changing some wiki links 
close #58
commit b3b08373 (2016-08-25): Changing some wiki links. close #58
closed 2016-08-25 by MaximeBaudette