OpenIPSL/OpenIPSL #67: Copyright assignment clause unclear or even invalid

dietmarw, 2016-07-01:
I've just read the [Copyright Assignment clause](https://github.com/SmarTS-Lab/OpenIPSL#copyright-assignment) and this sentence does not make much sense:
> "I hereby assign copyright of my contributions to the iPSL, to be licensed under the same terms as the rest of the iPSL."

Who are what is "the iPSL" supposed to be. The library is called OpenIPSL but you can not assign a copyright to a library but to some legal entity. 

I have a alternative proposal. Have a look at https://www.clahub.com/ and maybe use that one. Seems much clearer.

lvanfretti, 2016-07-03:
We didn't fix this when we forked, we'll take a look at your advise! @MaximeBaudette

MaximeBaudette, 2016-07-04:
@dietmarw 
The CLAHub idea seems really great, I will look into it to add it to the repository. 
I agree that the copyright assignation is not really clear, but our problem is that we have no legal entity that we can use for the copyright of the library, would have been good if we could have used SmarTS Lab, but our research group is no legal entity :-/

MaximeBaudette, 2016-07-25:
@dietmarw 
Do you think we could stipulate the following: 
> "I hereby assign the copyright of my contribution to the original authors of the OpenIPSL, to be licensed under the same terms as the rest of the OpenIPSL"

That's the only easy fix I see.

dietmarw, 2016-07-25:
1. I'm not a lawyer ;-)
2. I think you would need to make explicit who those original authors are.
3. You still need to find a way to keep record of the assignments.

MaximeBaudette, 2016-07-25:
> I'm not a lawyer ;-)
Me neither :-/
> I think you would need to make explicit who those original authors are.
Ok !
> You still need to find a way to keep record of the assignments.
Yeah, but your CLAhub solves this issue, right?  (I haven't implemented it yet, cause I want to fix the disclaimer first :-o)

dietmarw, 2016-07-25:
> Yeah, but your CLAhub solves this issue, right? (I haven't implemented it yet, cause I want to fix the disclaimer first :-o)
Yes exactly. I was not sure if you still wanted to use CLAhub or simply change the notice.

dietmarw, 2020-10-14:
the clahub.com service no longer exists. I've now removed the webhook from the settings. As an alternative I can set up https://cla-assistant.io/ which seems to be more robust and also the MSL is using. 
@MaximeBaudette Do you still have the actual CLA statement that you used for clahub.com?

dietmarw, 2020-12-11:
I found the CLA. It actually was just a section in the README and was removed in c26fd99d128f01cb091542165e80c4f9d40ab7e8 
So might just aswell create a new one from scratch.

dietmarw, 2021-01-28:
@lvanfretti You might want to discuss this in your team what you need in the CLA. Especially with a potential change of license the CLA is more important for the future than ever before.

marcelofcastro, 2021-11-07:
@dietmarw and @lvanfretti 

I was checking the license agreement for the Building's library and I noticed they add a paragraph in the end of the BSD clause 3 for contributors. I thought about adding the same paragraph in the end of the license, like the one below:

Contributor Agreement:
You are under no obligation whatsoever to provide any bug fixes, patches, or upgrades to the features, functionality or performance of the source code ("ENHANCEMENTS") to anyone; however, if you choose to make your ENHANCEMENTS available either publicly, or directly to Luigi Vanfretti, without imposing a separate written license agreement for such ENHANCEMENTS, then you hereby grant the following license: a non-exclusive, royalty-free perpetual license to install, use, modify, prepare derivative works, incorporate into other computer software, distribute, and sublicense such enhancements or derivative works thereof, in binary and source code form.

lvanfretti, 2021-11-07:
@marcelofcastro this would work fine for me, but let's see what @dietmarw thinks about it.

dietmarw, 2021-11-09:
By adding any more text to the original license text you would be basically creating a new (and no-longer OSI approved) variant of the BSD license. So I would stay away from such modifications. **But** the text itself is a very good template for a CLA that people need to (digital) sign when contributing to the project.

marcelofcastro, 2021-11-09:
Okay, so what about separating them? 

We have the BSD license in one file so people can look at that. In the contribution section we can make a link to this paragraph saying that, if they want the enhancements to be on OpenIPSL, they are acknowledging that paragraph and from now on the enhancement will be under OpenIPSL's BSD license. 

It might be a way of not modifying the BSD right?

dietmarw, 2021-11-09:
You simply put the orginal BSD3 license file in the repository and activate the https://cla-assistant.io/ with the other text. Still needs to be adapted though.

lvanfretti, 2021-11-09:
@marcelofcastro I looked at what Dietmar mentioned and he is right, we cannot make any modifications to the BSD license text, that modifies the license and it is no longer in the list from OSI: https://opensource.org/licenses

That will make it a "weird" thing that some people, like government agencies funding work, are not familiar with and will make exception for, which is the issue that we want to deal with for future project support.

What I suggest is that we separate the license from the copyright license agreement.
- Within the library documentation, we can list and include both.
- Within the repository, we can list and include both.

marcelofcastro, 2021-11-09:
I suggested the inclusion of that paragraph because buildings library has done it. And they continue to collaborate with government agencies and so on. It looked like a nice way to include the CLA. 

Anyway, we can go as @dietmarw suggested. I will include the modified version of the BSD3 (modified because we need to put the info from the library on the text like name of the product, year and who's the copyright owner) in the resource section of the library and then Dietmar can activate the CLA assistant. I don't have any idea how it works so I don't know what should be adapted...

lvanfretti, 2021-11-09:
@marcelofcastro one would assume that "What’s good for the goose is good for the gander," but remember the Buildings library is from LBNL, a big national lab, and not our rickety team of outcasts.

What I want to avoid is an application reviewer finding an excuse to reject an application due to a technicality... you know how people is by now, they won't treat us fairly compared to some big fish team of a pseudo-beautiful mind academic.

I agree with your approach.

I have activated the CLA assistant for the organization. 
[image]

Now, can you take a look at how to do it properly here: https://github.com/cla-assistant/github-action

closed 2022-03-11 by marcelofcastro