OpenIPSL/OpenIPSL #76: Check Compatibility with Wolfram SysModeler

MaximeBaudette, 2016-10-19:


tinrabuzin, 2016-10-19:
@MaximeBaudette @lvanfretti 

So, it seems like they fixed the issue we had with the complex numbers. However, the problem now is in the if statements in the line model that implement line opening. If I remove them, then the examples can be simulated. 

Nevertheless, for any of the more complex systems I get some compiler errors. If I'm not mistaken, the compiler complains about the too long paths to some of the cpp files that it is generating.

lvanfretti, 2016-10-19:
Hmmm, I think we should think about how to model the faults better and take
them outside the line... but I need to think about it.

Regarding the paths, please move where you put the OpenIPSL to a top layer
in your drive. I have had that problem with OM also.

Luigi

tinrabuzin, 2016-10-19:
@lvanfretti 

The compiler is complaining about the paths of c++ source files in the SystemModeler's directory so I think moving OpenIPSL won't help. And, as I said, I'm not sure that is in fact the problem. 

However, I did manage to run some of the larger models (Two_Area_PSSE, AKD, SevenBus).

tinrabuzin, 2016-10-21:
I think I found what is the problem. When SystemModeler generates the c++ files, there are several lines with function `solver_linear_system_scalar` and in the argument of this function there is a string with 36049  characters. That's what the compiler is complaining about (my VC 2015 compiler).

lvanfretti, 2020-04-23:
There's nobody to work on this, so I'm closing it.

closed 2020-04-23 by lvanfretti