OpenIPSL/OpenIPSL #81: Missing PSAT Models

MaximeBaudette, 2016-11-16:
This issue can be used to track the list of missing PSAT models: 

### Loads

- [ ]  Thermostatically Controlled Load

### Machines

- [ ] Synchronous Machine order V, Type 3
- [ ] Synchronous Machine order VIII

### Controls

- [ ] PSS Type I
- [ ] PSS Type III
- [ ] PSS Type IV
- [ ] PSS Type V
- [ ] Secondary Voltage Control
- [ ] Power Oscillation Damper

### FACTS

- [ ] SVC Type I
- [ ] SVC Type II
- [ ] SSSC Constant Voltage
- [ ] SSSC Constant Reactance
- [ ] SSSC Constant Power Flow
- [ ] UPFC Constant Voltage
- [ ] UPFC Constant Poweer

### HVDC

- [ ] HVDC

### Miscs

- [ ] Dynamic Shaft

dietmarw, 2020-10-19:
PSS Type I and III and Thermostatically  Controlled Load have been added in #232 from the FOSSEE1 branch

dietmarw, 2026-01-15:
I think PSAT has become a very low priority. So we might just aswell close this issue as "won't fix", right @lvanfretti ?

lvanfretti, 2026-01-19:
@dietmarw the issue is that I don't really have the resources to do these implementations.
I think we should keep this issue open so that people can see it, and perhaps by miracle, contribute. Or perhaps there is a better place to put this information?

dietmarw, 2026-01-20:
OK, in that case just leave this open.

lvanfretti, 2026-10-31:
**TL;DR.** This list is replaced by #437, which carries over the checklist and the comments posted here and orders the PSAT work in five topics (debug, integrate, validate, develop, the 2026 verification record) by what unblocks PSAT's own test cases. Closing this one so there is a single list to maintain.

**Key points**

- Everything from this issue is in #437 under "Carried over from #81": the thermostatically controlled load and PSS types I and III are done (#232) and verified against PSAT in 2026; the rest is still open and placed in its topic.
- Since 2026-09 twenty-nine PSAT models were verified against PSAT 2.1.11 with the PSAT-2-Modelica tool chain and twenty pull requests fixed what it found (#403 to #430); ten more PSAT models exist in the library and were never checked.
- The new list says what each missing model unblocks, so a contributor can pick by effect.

Generated with Claude Code

closed 2026-10-31 by lvanfretti