OpenIPSL/OpenIPSL #82: Missing PSSE Models

MaximeBaudette, 2016-11-16:
This issue can be used to track the list of missing PSSE models: 

### Loads

- [ ]  LDFRZN
- [ ] LDSHBL

### Machines

- [ ] CSVGN5

### Controls

- [ ] BBSEX1
- [ ] IVOEX
- [ ] ESAC5A
- [ ] EXELI
- [ ] EX2000
- [ ] GAST2A
- [ ] BBGOV1

### FACTS

- [ ] CSTCNT

### HVDC

- [ ] CDC6T
- [ ] CDCABT

### Miscs

- [ ] DPDTR1
- [ ] TIOCR1

### Wind

- [ ] WT1G1
- [ ] WT3G2
- [ ] WT3T1
- [ ] WT12T1

meliaa4, 2017-03-06:
Where  can I find wind modes ?

lvanfretti, 2020-05-21:
@marcelofcastro let's create a new issue with our new priorities. This issue refers to the models necessary to model the full Nordic grid model. Since we are not supported to do that work, we need to create another list that is aligned with the current efforts.

closed 2020-05-21 by lvanfretti