OpenIPSL/OpenIPSL #89: Various copyright statements still present in the development version

dietmarw, 2017-02-21:
I though the reason for the new script that adds the copyright was to clean out the development version of all the duplicates. But the latest (versioned) development version is still full with this "noise".  Is that on purpose or simply oversight?

MaximeBaudette, 2017-02-21:
@dietmarw 
Good remark !
Actually, now that I think about it, I would remove the noise alltogether and just place a notice on the top layer Modelica package... What an irony you are probably thinking!

dietmarw, 2017-02-21:
Yes, that idea does sound familiar somewhat. :smirk:

dietmarw, 2017-02-22:
Where you thinking of getting also rid of the `` place holder at the same time? Or do you think it should stay for later release processing?

MaximeBaudette, 2017-02-22:
Actually, if we were moving to removing all of them, I'd even remove this I think.

MaximeBaudette, 2017-02-22:
Anyhow, now that I prepared a script to update the annotations, it's easy to use to remove all of them also :smile:

dietmarw, 2017-02-22:
That's true, since you don't actually look out for that place holder at all.

MaximeBaudette, 2017-02-22:
Let's keep quiet on my quick n' dirty fixes :stuck_out_tongue_winking_eye:

dietmarw, 2017-02-28:
As part of #91, I've now added a copyright class and removed manually the local copyright statements whenever I touched a class. It would perhaps be a good idea to run your script after the merge of #91 in order to remove the remaining ones. I didn't want to touch it since it is hard coded for your machine @MaximeBaudette

MaximeBaudette, 2017-02-28:
Oh you're right.... :disappointed: 
Terrible! I really should fix that

janlav, 2017-07-07:
@MaximeBaudette is this addressed in your latest contributions?

MaximeBaudette, 2017-07-07:
Yes I fixed it

closed 2017-07-07 by MaximeBaudette