OpenIPSL/OpenIPSL #92: New dynamica power display

dietmarw, 2017-02-23:
I've created a new way of the power line display and wanted to get some feedback if this is fine for you.

I got rid of the arrows and replaced it with indicators instead. Also there won't be any negative values in the display but just flipping direction indicators.

Here is how it looks like in action. Left the new way I like to implement and to the right the old way:

[image]

MaximeBaudette, 2017-02-23:
It looks very good!

tinrabuzin, 2017-02-23:
@dietmarw I like it! :)

dietmarw, 2017-02-23:
OK in that case I'll continue with this. I leave this ticket open until the actual PR is merged.

dietmarw, 2017-02-23:
I've implemented it as par of #91

dietmarw, 2017-02-24:
Here the next improved generation that also allows flipping:
[image]

dietmarw, 2019-02-28:
Should have been closed long time ago.

closed 2019-02-28 by dietmarw