OpenIPSL/OpenIPSL #93: Wrong name of ShuntR?

dietmarw, 2017-02-27:
When cleaning up the icons I stumbled upon `OpenIPSL.Electrical.Banks.PwShuntR`
the description, the equations and the icon suggests this is a shunt L and not shunt R. 
Am I missing something?

tinrabuzin, 2017-02-27:
@dietmarw It seems like you're right, i.e. by looking at the equations it seems like it's shunt reactor. But that model was developed by other project partners so I don't remember seeing anyone using it in "our" models. Moreover, we have shunt B, C and R (which is L actually). So probably they included it because you can specify shunts in different ways, which means that we should maybe recode one shunt component with different parameter specifications...

dietmarw, 2017-02-27:
Yes having one with parameters would be an improvement. I will not change the name right now and we can leave this ticket to deal with it.

dietmarw, 2018-11-15:
My students just wanted to *use* that model and it turns out it is actually faulty and does all kinds of things but not what it should. 

We solved the issue by using either `PwShunt` or `Simulink.PwShunt` instead. The former is still causing a division by zero which should be looked at. See also #174.

So I'm for removing `PwShuntR` and `PwShuntC` given the `PwShunt` issue #174 is resolved.

marcelofcastro, 2019-07-29:
@dietmarw are you going to remove PwShuntC and PwShuntR from the library since issue #174 is solved?

commit 304e2ea0 (2019-08-23): Merge pull request #191 from dietmarw/t93-removeShuntModels

Fixes #93 by removing superfluous shunt models
commit 4b025c97 (2019-08-23): Fixes #93 by removing superfluous shunt models

`OpenIPSL.Electrical.Banks.PwShunt` simply fulfils all the requirements needed. So `PwShuntR`, `PwShuntC` and also `PwShuntB` have the same functionality with different implementation. Something that is highly confusing for the end-user.
closed 2019-08-23 by marcelofcastro