OpenIPSL/OpenIPSL #97: Generator Interface and Plant Model

tinrabuzin, 2017-03-01:
I just noticed that now we have a generator model in the `Interfaces` package. I started thinking about the plant model where you could select different parts of the plant model from a dropdown menu to avoid connections and a mess that comes with that. So, @dietmarw I'd like your opinion on the implementation. Do you think what I did makes sense? You can find the plant model in the "PlantModel" branch under `Electrical.Machines.PSSE`.

dietmarw, 2017-03-01:
Sorry but I can't find **any** model called `PlantModel`

tinrabuzin, 2017-03-01:
That's because the name of the model is `Plant` and not `PlantModel`. :) Sorry.

tinrabuzin, 2017-03-01:
Also, if you go to GENSAL example you can see how it would work. In Dymola at least...

tinrabuzin, 2017-03-01:
And yes, as I wrote in the first post, the model is in the "PlantModel" branch.

dietmarw, 2017-03-01:
OK now I see. Sure that's nice but since it is not yet supported in OMEdit, i would propably be careful as to how to introduce this.

dietmarw, 2019-02-06:
PR #157 implements the new structure without relying on replaceable support. I.e., there is an example "GEN" that demonstrates the use but the existing models are not touched so they stay usable with OMEdit.

dietmarw, 2019-02-28:
This was fixed as part of #177.

closed 2019-02-28 by dietmarw