OpenIPSL/OpenIPSL #102: simulation Wind turbine in PSSE

meliaa4, 2017-03-06:
I really need model of Wind turbine in PSSE. Can you help me! Thanks

tinrabuzin, 2017-03-06:
It would be nice to know who you are...

fran-jo, 2017-03-06:
And, exactly, which Wind Turbine model do you need!

meliaa4, 2017-03-07:
any model

meliaa4, 2017-03-07:
My name is Nguyen Huu Hao. I'm a student and studying in Viet Nam.

lvanfretti, 2017-03-08:
@meliaa4 I think you are a little lost.
Perhaps your Professor/Supervisor asked you to do this, or you are doing it out of your interest, but the question that you are asking doesn't make much sense for this repository.

Firstly, we do not provide any tools or support on how to simulate wind turbines using PSS/E.
Our library is to carry out power system simulations using Modelica-language-compliant tools, such as OpenModelica. A description of our library is provided here:
http://www.sciencedirect.com/science/article/pii/S2352711016300097

OpenModelica installers for multi-platform:
Windows: https://www.openmodelica.org/download/download-windows
Linux: https://www.openmodelica.org/download/download-linux
Mac: https://www.openmodelica.org/download/download-mac

In order to start working with our library, please go through the material in the following link, which is the most recent Tutorial/Overview that we have prepared:
https://github.com/SmarTS-Lab/OpenIPSL/releases/tag/Tuto_ModProd_2017

Having said that, we have re-implemented and validated against "reference tools", many of the models that are in our library. 
To simulate a wind-turbine-generator system, you can use some of our already available models, that you can find here:
https://github.com/SmarTS-Lab/OpenIPSL/tree/master/OpenIPSL/Electrical/Wind

In the PSSE folder, you will find several models that have been implemented and checked against PSSE, to understand the model, you would first need to get a copy of the documentation PSS/E here: http://w3.usa.siemens.com/smartgrid/us/en/transmission-grid/products/grid-analysis-tools/transmission-system-planning/Pages/University-Order.aspx

Please make sure you are completely familiar with the materials cited above before you go forward.

I will now close this thread, as it is not really an issue related to OpenIPSL at the time being.

closed 2017-03-08 by lvanfretti