OpenIPSL/OpenIPSL #107: PSAT generator models don't work correctly for non-nominal systemfrequency

dietmarw, 2017-03-09:
I found that the way  the electrical power is coupled with the mechanical power in `OpenIPSL.Electrical.Machines.PSAT.BaseClasses.baseMachine`:

```mo
 der(w) = (pm - pe - d*(w - 1))/m;
```
means that a generator produces more electrical power that it mechanically receives if the frequency is lower than the nominal, e.g., overloaded island grid. 

Are there alternative models that take this into account or all generator models not respecting the energy balance for non-nominal frequencies?

tinrabuzin, 2017-03-09:
I'm not sure if I understood your question correctly, but if you're referring to de damping coefficient "d" in this equation, that is a correct behaviour. This damping torque comes from the damper windings in the generator that improve the stability of the system during swings. 

I think this is a simplified way of modelling the damper windings since we're looking only at the 50Hz components in these kinds of models that we have here. Also, if I remember correctly, the contribution from this damping should be too big...

dietmarw, 2017-03-09:
No the issue is that independent of the damping (actually set to 0) for an overloaded islanded system the generator model generates more electrical power than it receives mechanical power. This is not a corner case but one of the most often asked scenarios when working with island or weak power systems. 

I don't doubt that PSSE has the same model but question is do you also have models that behave physically correct in addition?

tinrabuzin, 2017-03-09:
I don't see why would generator produce more electrical power than mechanical power (in the steady state)? Ofc, during the transients it would provide more power (stored in the rotating masses in the generator), but once the frequency settles down (to any value) the equilibrium would be restored.

dietmarw, 2017-03-09:
The why is simply because with `d=0` you end up with:

```mo
der(w) = (pm - pe)/m;
```
`pm` is constant, so is `m` and with a falling frequency this leads to `pe` becoming larger than `pm`. 
There is no restriction or safeguard present to avoid this. Well in this base model anyway. Hence my question if there is also a different implementation that does not suddenly creates power from nowhere :smirk: 

Here an example of an grid that loses connection to an infinite bus and starts to island with too little production available:
[image]

tinrabuzin, 2017-03-09:
The idea is that the power doesn't come out of nowhere. It comes from the "m" here which represents generator intertia. The thing is that eventually after the islanding, the governor would kick in, and start producing more power which would then bring the `der(w)` to zero and the intertia power (which you think is coming from nowhere) would go down. Your problem here is that there is no turbine governor model, so you should definitely add it. 

So, to sum it up more coherently, in real life you would have the same jump in electric power produced as you have here (coming from the inertia), it's just that the generator would be tripped if the frequency goes as low as yours did, or the governor would kick in and increase the produced mechanical power.

dietmarw, 2017-03-10:
@tinrabuzin You are unfortunately confusing a control system with a physical system. I don't blame you since most of the models in OpenIPSL are basically control system based and not physically. 
After the generator is islanding the **generator** can not produce more power as long as the supplied mechanical power is not changed. But this is not something that the generator does but the turbine governor changes the amount of mechanical power supplied to the generator. That is if you have more mechanical power available. Which is dependent on the operation point. So in my case I have a generator already at its maximum before islanding. So there is no additional power available and hence the frequency goes down. And this exactly the scenario that is of interest. How quickly does the frequency change when the grid is overloaded and what amount of loads need to be disconnected. 

So for such an scenario it does not help that the generator model suddenly produces electrical power out of nowhere (I never said there would be power coming out of nowhere but actually the opposite). Of course there are things like generator tripping on under frequency etc. but that is again a thing of the control system. At what frequency this occurs is also up to the control system. The generator model should not care about control logic but behave like a physically generator. 

Also the energy stored in the inertia of the generator would never cause the power to jump. It just slows down the frequency decent. So the energy is used to continue to provide the previous power. I've attached a single simplified model that shows the behaviour using torque sources (could be converted to power sources but would not make a difference. Especially since a hydro turbine is actually a torque source). 
[EquivalentPowerflow.mo.zip](https://github.com/SmarTS-Lab/OpenIPSL/files/833284/EquivalentPowerflow.mo.zip)

**Back to my original question:** Are there any generator models present in OpenIPSL that behave like a physical generator and not like a generator control system that is only valid for nominal frequency?

tinrabuzin, 2017-03-10:
@dietmarw Nope, I am not confusing control system with a physical system (that would be embarrassing). I was just explaining what would happen in a real power plant. Also, I was saying that what you get in the simulations that you were showing is exactly what should happen both in real life and simulations.

Check the slide number 5 in the following link [GE - Frequency Response](http://web.mit.edu/windenergy/windweek/Presentations/GE%20Impact%20of%20Frequency%20Responsive%20Wind%20Plant%20Controls%20Pres%20and%20Paper.pdf). The same thing that you show in your model is shown there, except that you lack the governor response because you did not include it in the model (or it is at its limits). 

Maybe a professor can solve the dilema here... @lvanfretti

tinrabuzin, 2017-03-10:
@dietmarw About your mechanical model, you should put another inertia between the generator and the power sensor. Then you'll get the same response as in our models.

dietmarw, 2017-03-10:
@tinrabuzin Thanks for the clarification. It was indeed the decoupled/split-up inertia that I was missing in my model and my train of thought. Well and the time-scale really. Simulating the model long enough until all the generators inertia stored energy has been depleted shows the (by me) expected power-flow. 

So this is then basically a non-issue.

closed 2017-03-10 by dietmarw