OpenIPSL/OpenIPSL #111: Typo in tutorial, Example 2

dietmarw, 2017-03-22:
I stumbled upon a parameter typo in the table given in the tutorial presentation. Since I don't have access to MS Power Point and LibreOffice would destroy some formatting I'm filing this as a ticket rather than a PR . 
The error is on slide 52 where this table is given:
[image]

Basically the values `Kw` and `Tw` are wrong. It should be:
 * `Kw = 9.5`
 * `Tw = 1.41`
These are also the correct values in the examples library.

Btw, should you at one point decide to switch the slides to LaTeX Beamer, then you will get PRs from me :smirk: .

dietmarw, 2017-04-19:
@lvanfretti I saw your newest tutorial slides still have this mistake. Also you should unify the slides for the current stable OpenModelica 1.11.0 release. The installation instructions are not in sync with the course content in this respect.

lvanfretti, 2017-04-19:
@dietmarw I got the chance of doing this in Texas A&M without much prior preparation, so I'm sorry I lost a bit of diligence (did it in the middle of a meeting).
Thank you very much for noticing the inconsistencies. I'm not sure I will have to update to 1.11.0 for tomorrow, def not in OSX, but I will try!

dietmarw, 2018-03-13:
Checking the [latest Tutorial](https://github.com/OpenIPSL/OpenIPSL/files/1253887/2017_08_.LVPT_OpenIPSLNAMUG_Prep.pdf) and it seems like it is fixed now.

closed 2018-03-13 by dietmarw