OpenIPSL/OpenIPSL #112: Default voltage values way out in PSAT models (and possible more)

dietmarw, 2017-03-23:
Just noticed that OpenIPSL's insistence on non SI base of k and M values instead of the real voltage and power values caused that many default values are too large by a factor of 1000. Reason is that whoever added the values in the first place thought of them (rightly so in my opinion) as Volts and are computational treated as kV. 

Example from OpenIPSL.Electrical.Branches.PSAT.TwoWindingTransformer`:

```mo
parameter Real Vn=400000 "Voltage rating (kV)"
```

There are two ways to solve this.
 1. (the quick and dirty and not modelica really): change all the default parameters to be of kV and M
 2. (the cleaner and you **REALLY** should consider this): Get rid of any implicit k and M values and treat parameters in there base units. Add displayUnits if you like them to be displayed otherwise but as is shown already in the current state even the original developers (who argue all the time that they only think in kV and M because of the domain of the library) managed to make a mess out of this. 

*The only reason this did not lead to disaster (yet) was that voltages were always used in a scaling fashion. So the error of 1000 was self-correcting. But I would not rely on this.

MaximeBaudette, 2017-03-28:
Interesting @dietmarw....
I can see why you would like to have Volts in the background, and I don't disagree with that. 
The only point I'd like to raise is that, in the end,  all quantities should be displayed in kV and MW, 
What ever Modelica uses in the background is not of big interest to me, but the display and input units should be kV (or p.u. :-P ), as it is the most familiar units. 

BTW: I think that if we go full option 2, it would quite easy to then later add better support for p.u. or am I wrong?

dietmarw, 2020-10-17:
@marcelofcastro   Did your work on the units already fix this?

marcelofcastro, 2020-10-19:
@dietmarw I will check if the units fix this issue

dietmarw, 2021-01-28:
@marcelofcastro I'm pretty sure this is fixed but still waiting on your check ;-)

marcelofcastro, 2021-02-01:
Yeah, I also think this issue is fixed. I've checked everything and everything seems ok in the main library. I think this issue can be closed, if you agree. If we find something later it's going to be an easy fix anyway...

marcelofcastro, 2021-02-08:
@dietmarw, I am closing this issue. As I said, if we find something later it'll be just a small fix. The issue is basically solved.

closed 2021-02-08 by marcelofcastro