OpenIPSL/OpenIPSL #119: Doubt with GGOV1 connection

maguilerac, 2017-04-18:
In some references I have seen that the governor output (output of the V Limiter block) is feeding both the main governor PID controller and the Rselect block.
However, I see that what is now being connected to the Rselect is the input of the V Limiter.

Is it correct? Should _connect(rSELECT.GovernorOutput, V.u)_ be replaced with _connect(rSELECT.GovernorOutput, V.y)_?

Reference:
Fig3-5 Dynamic Models for Turbine-Governors in Power System Studies (IEEE 2013)

MaximeBaudette, 2017-04-18:
@maguilerac You know you can add pictures to show the inconsistencies between the two.

maguilerac, 2017-04-18:
@MaximeBaudette Done!

Pictures for comparisson:

[image]
[image]

MaximeBaudette, 2017-04-19:
@maguilerac 
I didn't find anything in the history of the model that would justify this discrepancy between the model sheet and its implementation in OpenIPSL. BTW it was @tinrabuzin who added the model to the library, maybe he has some input. 
If not I suggest you prepare some tests and check the behavior of the model, before and after the fix.

lvanfretti, 2020-05-21:
Nobody followed up, I'm closing the issue.

marcelofcastro, 2020-05-21:
@lvanfretti I am following up on this. I am fixing GGOV

lvanfretti, 2020-05-21:
@marcelofcastro then assign yourself to what you are doing and make a note it the Issue.
I don't want any open issue without someone taking ownership of it. Otherwise, we are closing it and marking it with "wont fix"

marcelofcastro, 2020-05-21:
I'm taking over this issue

marcelofcastro, 2020-11-25:
Issue solved with PR #254

closed 2020-11-25 by marcelofcastro