OpenIPSL/OpenIPSL #138: Fix Dockerfile

MaximeBaudette, 2017-08-28:
I added a new account on Dockerhub to host the image for CI testing and configured it to automatically build the image based on the docker file hosted in the repository.  This is hosted in the "addDockerFile" branch. 

The generated docker image fails to run OMpython properly. 

Error to be investigated further.

MaximeBaudette, 2017-08-29:
@dietmarw 
For the first implementation of the CI, @tinrabuzin built a docker image that we host on a dockerhub account. This image was never updated and I was hoping to add a dockerfile to keep it in sync with the development of OpenModelica. 
I found an example file on the  OMWebbook repository, but didn't manage to get it working :-/ 
If you got any experience with docker+OM+OMpython, please let me know if you can help me fix this thing. 

I kind of gave up on updating OMpython on my personal computer, cause every time I face issues installing it :-S

dietmarw, 2017-08-29:
Have a look at my Docker file that I use for OM+OMPython+JupyterNotebook:
https://github.com/dietmarw/docker-notebook/blob/master/Dockerfile

commit 81fbffac (2017-11-01): Restore container tag to master
Fix #138
closed 2017-11-02 by tinrabuzin