OpenIPSL/OpenIPSL #148: Old, unused PSAT machine models in folders?

janlav, 2017-10-19:
Inside OpenIPSL\OpenIPSL\Electrical\Machines\PSAT\ there exist *.mo-files, which are loaded into OpenIPSL, but also redundant and/or old models in folders, f ex "FifthOrder". Do these serve any purpose and/or should they be deleted?

dietmarw, 2017-10-19:
To me it looks like a left over from reorganisation. Probably run a diff on the files in the folder and the one on the level above. They are not "linked" from package.order so they would not be shown.

MaximeBaudette, 2017-10-19:
I'm preparing a new release soon, I will use the occasion to clean this up. (Usually I load the library in dymola, delete the files in the repo and redo a `save as` so that old files that we forgot about just get removed)

dietmarw, 2017-10-19:
Oh dear and that explains the umpty changes to OpenIPSL in a single "clean up" commit. ;-)

I might suggest you do the following:
 1. Open in Dymola
 2. Remove all files
 3. Save in Dymola
 4. Only mark **deleted** files for commit. Don't commit any changed files.
 5. Commit. 
 6. Close Dymola
 7. Do a `git reset --hard HEAD` 
 8. Open in Dymola in order to check if everything still is fine or if some files needs still changing (`package.order`) would be a candidate. 

All this can already be done now.

lvanfretti, 2017-10-19:
@MaximeBaudette Maxime
The fifth order model from PSAT is necessary, even if it has not been updated, keep it.
I suggest you look through the PSAT manual to see the models that should be there, before you delete something.
If the model is not supported in the current library, we can put a mark in the code and in the icon for the user to know that the model needs maintenance.

Thank you for cleaning up the library!!!

Luigi

dietmarw, 2017-10-19:
I think the problem here is that there is something that is on the file system but not shown by the modelling tool. It is quite dangerous to keep stuff "around" like that by that method. I'd suggest to either create a dedicated package to put those models which is also shown in the modelling tool instead.

lvanfretti, 2017-10-19:
I am not sure if I understand, but the model is available in the original
modeling tool.

I don’t mind which ever approach is used, but since the models where
developed and used to work at some point, they probably only need
maintenance instead of full reimplementation.
I just don’t want to have to re-implement any model again, which is listed
in the original software (PSAT), just because we have not maintained it.

MaximeBaudette, 2017-10-20:
@lvanfretti 
We are not talking about deleting models Luigi, only deleting unused files. 
If you remember some models have been reworked (like the PSAT machines), and the packaged cleaned up. But I looked at the repo and it seems that files that were deleted within Dymola have not been deleted on disk, and hence from the repo. These files are just old copies:
[image]

PR #151 (2017-10-20): Remove duplicate files
Fix #148
commit 120c851c (2017-10-20): Remove duplicate files
Fix #148
closed 2017-10-20 by janlav