OpenIPSL/OpenIPSL #150: Possible bug in ESST4B

janlav, 2017-10-20:
I noticed on [https://libraries.openmodelica.org/branches/master/OpenIPSL/OpenIPSL.html](OMs OpenIPSL test) that [Examples/.../ESST4B](https://github.com/OpenIPSL/OpenIPSL/blob/master/OpenIPSL/Examples/Controls/PSSE/ES/ESST4B.mo) didn't compile and saw some strange things:

1.  The test compiles and runs under Dymola but both limiters are at-limit all of the time so that the only part that does something to the excitation voltage is the "Rectifier Commutation Voltage Drop" , it seems to be bad test of the system. Anyone agree?
2. The voltage is 1.56 pu at the generator, but the PF-solution say that it should be unity! This is likely why OM fails.
3. The sensed complex voltage and current seems to be wrong:
 ` V_T = Gen_terminal.vr + j*Gen_terminal.vi;`
  `I_T = Gen_terminal.ir - Bus.ir + j*(Gen_terminal.ii - Bus.ii);`
This seems to be **doubling** the real and imaginary components of the current as it is used in the ESST4B-example.

Both the example and model should be investigated.

marcelofcastro, 2020-05-21:
I'm taking over this issue

dietmarw, 2020-10-16:
I just ran the example and can confirm 3. . The OpenModelica issue (2.) seems no longer valid as I get the the expected 1 pu voltage value.

dietmarw, 2020-10-16:
Thinking about 3. a bit  ... the equation for I_T is simply wrong. It should be:
```diff
- I_T = Gen_terminal.ir - Bus.ir + j*(Gen_terminal.ii - Bus.ii);
+ I_T = Gen_terminal.ir + j*Gen_terminal.ii;
```

marcelofcastro, 2020-10-16:
So, (1.) and (2.) were basically related to the initialization of the exciter and due to some connections that were wrong. This is solved already. About (3.), @dietmarw is right and we just need to replace the equation.

dietmarw, 2020-10-16:
OK I'll provide a fix as soon as #228  is merged.

PR #230 (2020-10-17): Correct calculation of I_T in ESSTB4
This fixes #150.

Best to wait with merging until #228 is merged.
commit 957d5c86 (2020-10-17): Fixes #150  by correcting I_T calculation.
closed 2020-10-17 by dietmarw