OpenIPSL/OpenIPSL #160: Error in simulating Tutorial examples with OpenIPSL v1.5.0

emesk, 2018-02-27:
I noticed while simulating Example 1 from tutorial section 
https://github.com/OpenIPSL/OpenIPSL/blob/master/ApplicationExamples/_Tutorial/Tutorial/Example_1/Example_1.mo
on the release package OpenIPSL v1.5.0 
https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0

i get the error.

`LOG_NLS | warning | <p>nonlinear system 318 fails: at t=0</p>`

I am using default solver settings in open Modelica.

MaximeBaudette, 2018-02-27:
I checked quickly and noticed that the model in question, and also Example 2 was not working at all in OM, but it was working 'kinda' fine in Dymola (the initialization was incorrect). 
After checking, it turns out we updated the models to input the angles from the powerflow results in degrees instead of radians (before some models used radians, others used degrees, all models should use degrees now). 
I updated the models Example 1 & 2, the initialization was fine in Dymola, but still not working in OM

MaximeBaudette, 2018-02-27:
I changed some attributes on the initialization in the machine following what OM was complaining about. I don't really understand why all of a sudden OM became more picky about the initialization of the model.
See: 
`Real e1q(start=e1q0, fixed=true)`
`Real e2q(start=e2q0, fixed=true)`

MaximeBaudette, 2018-02-27:
@dietmarw 
Any input about the `fixed = true` statements? 
Should we use it in more places for proper initialization? 
Also, any idea why it stopped working in OM? I checked the commit history and there is nothing shocking about these models that would have broken them.

dietmarw, 2018-03-01:
To make models more robust you should always fully specify the initial conditions if known and tell the tool to use them (i.e., `fixed=true`) and not just take them as a guess (i.e., `fixed = false`) . Dymola also warns you each time about this (also the same places as OMEdit does) but in contrast to OMEdit adds the `fixed=true` part automatically.

So it might be a good idea to go through all examples and fix this.

dietmarw, 2018-03-01:
As to why OMC changed its behaviour maybe @sjoelund might shine some light on.

sjoelund, 2018-03-01:
The behavior when not specifying initial conditions is a bit random and whenever we do anything in the compiler that changes the sorting of equations (perhaps fit efficiency reasons storing something in a sorted tree instead of an unsorted list) we might make a different choice than we did before. We also frequently end up with a different set of initial conditions where we have a choice to set fixed=true. It's the same thing with start values for nonlinear systems; in different OpenModelica versions different iteration variables might be chosen and a start value that was previously provided might be ignored and a variable without start-value might be used instead..

MaximeBaudette, 2018-03-16:
@sjoelund Sorry for the delayed answer, and thanks for the input. 
I guess we should review our initialization procedure to make them more robust! 
The issue, I guess, is that I mostly work in Dymola, and it seems it's always been very forgiving on the initialization questions.

MaximeBaudette, 2018-03-16:
Change of title to "Review Initialization to comply with OM"

Old title: "Error in simulating Tutorial examples with OpenIPSL v1.5.0"

closed 2022-05-31 by dietmarw