OpenIPSL/OpenIPSL #167: Add phase shifts (clock numbers) to transformer models

dietmarw, 2018-05-31:
None of the current transformer models takes the phase-group/clock-number into account. This make comparison hard with tools like PowerFactory which offer these settings.

lvanfretti, 2018-06-06:
@dietmarw I'm not familiar with that feature of PowerFactory. Phase shifting transformers corresponding to PSAT are [here](https://github.com/OpenIPSL/OpenIPSL/tree/master/OpenIPSL/Electrical/Branches/PSAT/PhaseShiftingTransformer).

[In this MSc thesis](https://repository.tudelft.nl/islandora/object/uuid:a179674c-7ffa-4909-85e6-caa8c44dcc11/datastream/OBJ/download), Harish implemented some PowerFactory models compatible with OpenIPSL; however, they have not been integrated to the library (I haven't got permission from the author yet). 
Maybe you can take a look at the thesis and make the enhancement. 

When @marcelofcastro joins, his focus is going to be mostly on tools used in North & South America; so we need collaborators that cover the European tools.

dietmarw, 2018-06-06:
I'm not sure if the existing ones are what we had in mind. Will have to look at it. But looking at the package https://github.com/OpenIPSL/OpenIPSL/tree/master/OpenIPSL/Electrical/Branches/PSAT it is confusing that there are two (undocumented as usual) packages
 * `PhaseShiftingTransformer`
 * PhaseShiftingTransformer

lvanfretti, 2018-06-06:
They might not be exactly what PowerFactory has, but surely very similar.
Yes, we need to clean up this...

dietmarw, 2018-06-06:
Yes this definitely needs to be cleaned up. Phase shift should be part of the normal two or three winding transformer and not a separate model. So a cleaner implementation is asked for.

lvanfretti, 2018-06-06:
@tinrabuzin I think you are right.
I'm having a call with Dietmar this afternoon about something else, so I'll try to figure out what he means.

dietmarw, 2018-06-06:
@tinrabuzin Yes you are right. The PSSE implementation actually has the `ANG1` parameter.

dietmarw, 2018-06-06:
So basically the PSAT version does not offer this. But now I also have learned about what a phase shifting transformer is. :-)

lvanfretti, 2018-06-06:
There should be some phase shifting transformer model in PSAT and OpenIPSL… Ahsan implemented for his thesis… might not be cleaned up and integrated though...