OpenIPSL/OpenIPSL #170: systemmodeler developers feedback/potential patch?

lvanfretti, 2018-06-13:
I had a telco with the people from Wolfram about the issues when using SystemModeler and the library.

Here is the feedback:
> In some models, for example OpenIPSL.Examples.Machines.PSSE.GENSAL, there is an equation, gENCLS.omega = 0.0. But gENCLS.omega also has a start attributed (start=0.0) which is enforced with fixed = true. This will there to be two different equations for the gENCLS.omega variable during initialization. We believe the intention here was to use der(omega)=0.0, rather than the equation omega=0.0. Please see the attached patch for this proposed fix.

Proposed Patch:
[GENCLS.patch.zip](https://github.com/OpenIPSL/OpenIPSL/files/2099993/GENCLS.patch.zip)

> diff --git a/OpenIPSL/Electrical/Machines/PSSE/GENCLS.mo b/OpenIPSL/Electrical/Machines/PSSE/GENCLS.mo
> index e92fbd39..963d4e56 100644
> --- a/OpenIPSL/Electrical/Machines/PSSE/GENCLS.mo
> +++ b/OpenIPSL/Electrical/Machines/PSSE/GENCLS.mo
> @@ -55,7 +55,7 @@ equation
>      der(omega) = (P_0/S_b - P - D*omega)/(2*H);
>    else
>      der(delta) = 0;
> -    omega = 0;
> +    der(omega) = 0;
>    end if;
>    // d-q voltage and current equations
>    der(eq) = 0 "Classical model assumes constant emf";

I'm afraid we might have more than 1 inconsistence of this type in the library which is breaking SystemModeler.

dietmarw, 2018-06-14:
There are definitely more. Even that very example causes Dymola to reports two more:

```
The iteration variable pwLine.n.vr has been selected to have the guess value constantLoad.vr0.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
0.9919433259086916, the start value of constantLoad.p.vr given as constantLoad.vr0.
1.0, the start value of LOAD.p.vr given as LOAD.V_0*cos(LOAD.angle_0*3.141592653589793/180).

The iteration variable pwLine.n.vi has been selected to have the guess value constantLoad.vi0.
However, the start value has been selected from a set of alias variables having conflicting start values of the same precedence.
-0.009977085118074672, the start value of constantLoad.p.vi given as constantLoad.vi0.
0.0, the start value of LOAD.p.vi given as LOAD.V_0*sin(LOAD.angle_0*3.141592653589793/180).
```

lvanfretti, 2020-05-21:
I'm giving up supporting SystemModeler, we don't have the bandwidth.

dietmarw, 2020-05-28:
This has nothing to do with SystemModeller but rather incorrect initialisation in general. Just like Dymola also supported. So this should not be closed but fixed since it will cause problems independent of the tool. I've also changed the title.

marcelofcastro, 2020-10-23:
So the conflict was very simple. Bus LOAD needed to have the same v0 and angle0 as the device Load in the SMIB examples. They were defined with different values although they were supposed to have the same value. In addition, the equation for GENCLS should, in fact, be `der(omega) = 0` if H = 0, instead of `omega=0`. I am preparing a PR to solve this issue.

marcelofcastro, 2020-10-23:
#229 will be solved with this PR as well.

PR #240 (2020-10-24): Fix gencls
This update should fix some initialization problems with SMIB system and the over-specified equation for omega. This might address issues #170 and #229 

(fixes #170 and fixes #229)
commit 11ed9c16 (2020-10-24): Fixes #170 by fixing equation for inf bus
closed 2020-10-24 by marcelofcastro