OpenIPSL/OpenIPSL #173: Inconsistent results from softPMU.

dgusain1, 2018-11-15:
Hi!

I am testing the softPMU component in IEEE9bus example. The results obtained are very inconsistent and do not make sense. The frequency plotted from generator is very different from frequency reported by softPMU at same generator terminal.

[image]

As you can see, the softPMU reports a negative frequency! Any reason as to why this may be happening?

dietmarw, 2018-11-15:
Could you share your model? It could be also an issue of the units and scaling. Easier to see if one can look at how you implemented the SoftPMU

dgusain1, 2018-11-15:
I picked up the model from Application Examples packaged with OpenIPSL. Im attaching it here. (Use the IEEE_9_Buses within the package.)
[IEEE9Total.zip](https://github.com/OpenIPSL/OpenIPSL/files/2584445/IEEE9Total.zip)

dietmarw, 2018-11-15:
Sorry the attached model is empty (basically):

```
model IEEE9_total
  extends IEEE9;
end IEEE9_total;
```

dgusain1, 2018-11-15:
Oh, my bad. Here is the model file.

[IEEE9.zip](https://github.com/OpenIPSL/OpenIPSL/files/2584512/IEEE9.zip)

dietmarw, 2018-11-15:
Sorry but the model IEEE_9_Buses is missing parameters and does not run. That way it is hard to help you. Could you supply a working (complete parameterised) example that also checks and simulates?

dietmarw, 2018-11-16:
I see now the problem and can verify that the results as generated by `softPMU` do not make sense. Looking at the internals the root of the problem is they way the reference angle is calculated using derivatives. This leads both to spikes and also delays (i.e., phase shifts).

I have a Master Thesis coming up in spring 2019 where I hope to get a better implementation that then most likely can replace the faulty (in my opinion) `softPMU` model.

lvanfretti, 2018-12-04:
Please see the usage example in this repo:
https://github.com/ALSETLab/2018_AmericanModelicaConf_PMUBasedIslanding

I haven’t had the time to commit this new version.
Maybe @Biswarup can do it?

See the paper for implementation approach to eliminate spikes using the smooth operator.

BiswarupM, 2018-12-04:
@dgusain1 - The softPMU model, the one you're using needs to be corrected it seems. Please make sure that the real and imaginary parts of the voltage are assigned properly. You can also use the model from the paper that Prof. Vanfretti mentioned above. 
Good luck!

lvanfretti, 2018-12-04:
> I had a look and the version in that publication (btw the equation (2) in the paper is wrong, but correct in the model) and I can not find any difference to the already existing (but misspelled) frequency calculation model: OpenIPSL.NonElectrical.Nonlinear.FrenquencyCalc.
> So I'm not sure what was solved in the paper and why the duplicate Modelica2018US_IslandedOperationControl.Essentials.FrequencyCalculationBlock was created. It doesn't do anything differently (and just as faulty) as the existing block.
> I tested it with the IEEE9 bus model and they give the exact same faulty frequency results.
> @BiswarupM Can you explain what was done here was was tried to be done?

I think Biswarup didn’t save it correctly, I don’t have the files myself; my computer crashed, and I can’t find a copy on his dropbox.

@BiswarupM,

Can you please update the repository?

Luigi

lvanfretti, 2018-12-04:
Actually, I’m wrong.
We used the homotopy and smooth operator in another paper for deadline with angle wrapping problems, not for frequency calculations.

I think the problem is that the modifications made were not saved correctly in the model in the repo, the implementation we used is this one:
https://github.com/ALSETLab/2018_AmericanModelicaConf_PMUBasedIslanding/blob/master/Modelica2018US_IslandedOperationControl/Essentials/FrequencyComputationBlock.bak-mo

In any case, Biswarup should upload the updated model.

Luigi

dietmarw, 2018-12-04:
OK but that model then contains the wrong equation (identical to (2) from the paper).
See also https://www.wolframalpha.com/input/?i=derivative+of+tan%5E-1(x(t)%2Fy(t))

lvanfretti, 2018-12-04:
> And does not solve any problem since it is just the equation rewrite of
> the block diagram from before 👎
>
> So it has the same problems as the existing component and again the
> question is why was it added?

Dietmar,
We need some sort of frequency estimate for certain controls, as explained
in the paper.
In the paper, we were trying to get the thing implemented in a way that we
don't get spikes.
It was added at the time because it is a nice way to package certain
simulation values "measurements", to be displayed and re-used.
Of course, if there is a mistake, it should be corrected.
Remember that most of the stuff is done by students, and even if I check
the plots, sometimes it is hard to find the errors.

The main problem to be solved is to put the right equation and
implementation so that one doesn't get switching in.
Unless you implement a frequency estimator (something I've planned to do
but don't have the manpower), this is a shortcut.

Luigi

-- 
Luigi Vanfretti, PhD, Associate Professor
ALSETLab.com <http://alsetlab.com/> - ECSE <https://www.ecse.rpi.edu/>, RPI
<http://rpi.edu/>, Troy, NY, USA <https://goo.gl/maps/mVowWWPLtL72>
E-mail: luigi.vanfretti@gmail.com
iPhone: +1 (518) 496 0196 - Skype: lvanfretti
Github <https://github.com/lvanfretti>  -  Google Scholar
<https://scholar.google.com/citations?user=_XI6Q6UAAAAJ&hl=en&oi=ao>  -
ORCID <https://orcid.org/0000-0002-4125-1055>  -  LinkedIn
<https://www.linkedin.com/in/vanfretti/>  -  Twitter
<https://twitter.com/lvanfretti>  -  Youtube
<https://www.youtube.com/user/lvanfretti>

BiswarupM, 2018-12-04:
Please see the real and imaginary parts are indicated other way around. In my AMC paper and the one Jan committed. So my model https://github.com/ALSETLab/2018_AmericanModelicaConf_PMUBasedIslanding/blob/master/Modelica2018US_IslandedOperationControl/Essentials/FrequencyComputationBlock.bak-mo
 is correct. There is nothing wrong

dietmarw, 2018-12-04:
Unfortunately from that model you can not determine what should be real and what should be imaginary. But anyway. It should be minus not plus on top.

BiswarupM, 2018-12-04:
@dgusain1 If you go here 
        2018_AmericanModelicaConf_PMUBasedIslanding/Modelica2018US_IslandedOperationControl/Network/Network_DL.mo
you can see how the block inputs are specified

SimulationSetups.Event_Islanding event_Islanding( f_DN(u = B6.p.vi, u1 = B6.p.vr))

@dietmarw @lvanfretti

BiswarupM, 2018-12-05:
Speed= 2*(Pi)*f
And when you plot them in same unit you get the following. 
[image]

dietmarw, 2018-12-05:
And that discontinuity is not acceptable for general purpose applications. So it needs to be fixed,

marcelofcastro, 2022-05-26:
I have investigated this issue and I've found several tiny issues with the frequency calculator. First of all, the frequency calculation should be `(Vr*der(Vi) - Vi*der(Vr))/ (Vr^2 + Vi^2)` as @dietmarw and @BiswarupM mentioned above. However, the block `FrenquencyCalc` had it `(Vi*der(Vr) - Vr*der(Vi))/ (Vr^2 + Vi^2)` and that needed to be fixed.

In addition, I had to do some other small adjustments to frequency estimate, since now we only use radians for calculations. Apparently, instead of dividing it by two times pi, the estimator was converting the estimate to degrees, which I think was also a tiny error that was fixed.

Furthermore, I've propagated the time constant associated with the derivative filter calculation so the user can adjust it in order to smooth the frequency estimation according to the application. I am attaching the comparison of the same estimate done in the IEEE9 bus using different time constants. The first image is the frequency estimate using a filtering time constant of 10ms.

[image]

This second image is using a filtering time constant of 50ms. Note that a larger time constant eliminates the sharp large peaks. 

[image]

I am attaching the IEEE 9-bus that can be used for testing the renewed softPMU that will be available in the next PR.

[IEEE_9_Buses.zip](https://github.com/OpenIPSL/OpenIPSL/files/8780654/IEEE_9_Buses.zip)

marcelofcastro, 2022-05-28:
Another thing that must be kept in mind @dietmarw is that the softPMU should not give you the exact frequency, since it is a measurement device supposed to estimate local frequency from a bus using the information from the phasor angle. 

Indeed we could use simply `der(angle)` or, as you suggested, `der(atan2(Vi/Vr))` but @BiswarupM and @lvanfretti opted for using a different way to estimate the frequency deviation, while avoiding divisions by zero and using filters to avoid many of the chattering that can be caused by a true derivative process (especially if we are talking about real measurements). In addition, this method they are proposing is way different from what is used in literature (the washout filter they mention) and it is also shown to be more robust. 

In summary, the softPMU is just a device used to reproduce frequency estimation using the phasor angle that is measured from a PMU. Therefore, there is no need for it to exactly calculate the local frequency and the model is also more "realistic" in terms of what is really implemented in PMUs for local frequency estimation. Therefore, users can benefit from such model by using a device that can represent such measurement and estimation procedure to properly design controllers, for instance.

dietmarw, 2022-05-30:
I just ran your IEEE9 bus example with `Ts = 0.05 s` and get the following result which is quite different to the one you are showing: 
[image]

I could not find where `Ts` is propagated to. Did you forget to push the latest change?

dietmarw, 2022-05-30:
I've fixed the propagation of Ts now and get the same result. I'm still wondering how useful a PMU device is that even after 1 second after an event still has an error of 0.2 Hz when changes of more than 0.1 Hz are already seen as sever. Could you give a pratical example where the application of a PMU would be fine with such a large margin of error?

dietmarw, 2022-05-31:
But that is the point, the softPMU is just a wrapper around the FrequencyCalc block and the level of error that generates (ignoring even the spikes) makes its practical use highly questionable. I just like to make sure that users don't get the impression that the softPMU will actually give an output of practical use.

closed 2022-05-31 by dietmarw

lvanfretti, 2022-05-31:
In the future I want to have @marcelofcastro or someone else implement the methods we studied with real world measurements data here: https://sites.ecse.rpi.edu/~vanfrl/documents/publications/conference/2022/CP207_Chetan_Derivatives_PESGM_lvrev.pdf

We may also want to try to use complex step derivative: https://mdolab.engin.umich.edu/wiki/guide-complex-step-derivative-approximation#:~:text=The%20complex-step%20derivative%20approximation%20is%20a%20very%20convenient,of%20a%20quantity%20calculated%20by%20an%20existing%20algorithm.

@dietmarw as you can see in the above paper, the real world instruments are very far from giving an ideal derivative, while the so-called washout filter approach used in PSS/E is very susceptible to chattering and other issues.

The current implementation is good enough for most practical cases, as long as you parametrize it correctly, at least it gives better results than what the de facto method can do. Anyway, we’ll add more methods for the user to choose from in the future.