OpenIPSL/OpenIPSL #174: PwShunt causes division by zero

dietmarw, 2018-11-15:
When using `Electrical.Banks.PwShunt` Dymola (at least) displays 
``` 
Model error - division by zero: (1.0) / (pwShunt.p.ir^2+pwShunt.p.ii^2) = (1) / (0)
```
but in the end succeeds.
This should be investigated.

dgusain1, 2019-02-20:
I have similar problems. But in my case, the simulation does not go through. Im using the Two area system from Application Examples.

marcelofcastro, 2019-05-27:
This problem occurs also when using `Electrical.Banks.PwShuntC` and `Electrical.Banks.PwShuntR` and it occurs during the initialization. When the variable is an iteration variable of the initialization problem and no start value is given, 0 is used and this is what is causing the problem. Some non-linear solvers in Dymola are able to deal with the problem (as it is shown below) but I don't know about other solvers and other software tools. @dgusain1 are you using Dymola? If yes, try to change the solver to _dassl_.

'Non-linear solver will attempt to handle this problem.'

One solution is to define an initial guess for the imaginary and real parts of the current, that is not zero. For instance, we can initialize these values with Modelica.Constants.eps, which is defined as:

> Biggest number such that 1.0 + eps = 1.0

Hence, the code written below would solve the error/warning.
`OpenIPSL.Interfaces.PwPin p(
ir(start=Modelica.Constants.eps), ii(start=Modelica.Constants.eps)
)` 

@dietmarw do you have a better solution for this?

dietmarw, 2019-06-20:
That looks like a good solution. Actually I'm going one step further and defining the non-zero start value (only a guess value after all) for `OpenIPSL.Interfaces.PwPin.ir|ii` itself. I don't think that will cause any problems and will on the contrary solve also other example models. PR is coming up.

marcelofcastro, 2019-07-01:
@dietmarw That means that this issue was solved, right?

commit 7f4d6538 (2019-07-01): Fixes #174 by providing a non-zero start value for ii and ir
commit 9eac7c28 (2019-07-01): Merge pull request #190 from dietmarw/t174_iniitNonZero

Fixes #174 by providing a non-zero start value for ii and ir
closed 2019-07-01 by lvanfretti