OpenIPSL/OpenIPSL #179: Update contact details for 2.0.0 release

dietmarw, 2019-02-07:
Currently many places link to @lvanfretti 's old email address as contact info. For the 2.0.0 release this should be updated. But instead of using another email address I think it would be good to link to a webpage that contains contact information. Question is which one, openipsl.org or alsetlab.github.io

Nothing urgent but should be done before the 2.0.0 release.

lvanfretti, 2019-02-07:
Agreed @dietmarw I guess openipsl.org would be the best fit as it pertains questions only to openipsl. @marcelofcastro can take care of this.

dietmarw, 2020-10-23:
I noticed that instead of pointing to openipsl.org the updates done recently simply update Luigi's email address. This is not ideal. 
The whole table itself should be moved into the revision section (see `PhaseShiftingTransformer` in  PR #236 as an example). 
But in general I would say the contact row is really superfluous. Simply pointing to (or just having) a proper `UsersGuide.Contact` class with all information on the developer team is enough and does not add all that noise  in all the models that also  gets outdated quicker than one thinks.

lvanfretti, 2020-10-23:
@dietmarw I agree, we should have the `UsersGuide.Contact` instead.

closed 2020-12-10 by marcelofcastro