OpenIPSL/OpenIPSL #186: Include PSAT PSS and load models from FOSSEE1 branch

dietmarw, 2019-02-28:
In order to check some boxes in #81 the FOSSEE1 branch provides the following models that could be merged(target location):

 - Examples.Loads.PSAT.ThermostatLoad
 - Examples.Loads.PSAT.BaseClasses.LoadBase (and adapt the other existing load tests for this)
 - Electrical.Loads.PSAT.ThermostatLoad
 - Electrical.Controls.PSAT.PSS.PSSTypeI 
 - Electrical.Controls.PSAT.PSS.PSSTypeIII

The OEL models are duplicates of the existing models.

The only thing left (despite the rearranging of course) is adding some documentation to the models and also having an example each for the PSS models.

@BiswarupM Do you still have the possibility to contribute here?

dietmarw, 2020-10-17:
@Biswarup I was wondering if you could simply add some short documentation to the models. I can then take care of adding them to the OpenIPSL library itself.

dietmarw, 2020-10-19:
Never mind @BiswarupM. I've added some references to the models in #232.

closed 2020-10-19 by dietmarw

BiswarupM, 2020-10-19:
No problem @Dietmar. I just saw it.
Thanks!

PR #232 (2020-10-19): Include PSAT PSS and load models from FOSSEE1 branch
This fixes #186 by including the new load  and PSS models as developed in the FOSSEE1 branch. At the same time the existing load examples and model have been cleaned up and also  properly named (conversion script is also updated).