OpenIPSL/OpenIPSL #189: Inclusion of discrete ULTC

dietmarw, 2019-06-19:
The current model of the ULTC is continuous only and not really very useful for that reason. A better discrete implementation was developed by Mohammed Ahsan Adib Murad. Now when trying to adapt his model to the current OpenIPSL structure i have the following question:

  The current continuous model is based on PSAT and hence placed under `OpenIPSL.Electrical.Branches.PSAT.ULTC_VoltageControl`. So is the discrete model also based on PSAT and should be placed side by side or should it be places in an entirely different location in the structure. E.g., ``OpenIPSL.Electrical.Branches.Transformers.ULTC` ?

ahsanKTH, 2019-06-19:
I agree with the location you suggested ('OpenIPSL.Electrical.Branches.Transformers.ULTC'). My only suggestion is to rename for example: 'OpenIPSL.Electrical.RegulatingTransformers.ULTC'. Then we can put all different regulating transformers into the package: RegulatingTransformers. Because in coming days we will add few more ULTC models (different discrete type) in the library.
Or Luigi can decide in which package to put the discrete ULTCs.

lvanfretti, 2019-06-19:
Hi,

No, the discontinuous model is not based on PSAT, is based on the paper from Hiskens.

So, I think it should have a different place in the structure.

In the library, depending on the "reference" upon which the model is based, we use sub-packages according to the reference, e.g.:
`OpenIPSL.Electrical.Branches.PSAT.x_branch_model_name`

So, to be consistent, unless the model is in PSS/E or in PSAT, we should create a new sub-package:
`OpenIPSL.Electrical.Branches.Generic.`

In the case of ULTCs, we can then do:
`OpenIPSL.Electrical.Branches.Generic.ULTC`

The documentation should include the reference from where the equations were pulled out from.

In this way we can generalize with other models, from wherever, where there is no way of checking against a simulation, such as in books or papers.

Luigi

dietmarw, 2019-06-19:
Generic sounds good. I wonder if it really is necessary to have an own ULTC package instead of having a "Transformer" package where we then can have the different transformer models (controlled and non-controlled).

As for documentation, I will be adding the PDF and the LaTeX sources in an improved (and fixed) way. You can already check them out in my branch: https://github.com/dietmarw/OpenIPSL/tree/ULTC/OpenIPSL/Resources/Documents/ULTC

dietmarw, 2019-06-24:
So I started with moving the classes of interest over into OpenIPSL. I've ended up adding three new classes: 

 * `OpenIPSL.Examples.Branches.Generic.ULTC` which represents the updated and cleaned up version of `Hisken.DAIS.Test_System_HiskenLoad_new`
 * `OpenIPSL.Electrical.Branches.Generic.ULTC` which is `Hisken.DAIS.TransformerDiscrete2` (no point in including the older work in progress models)
 * `OpenIPSL.Electrical.Loads.Generic.Load_Hisken` which is `Hisken.DAIS.Load_Hisken_new`

I'm still undecided if `Load_Hisken` is really a good name. Also is it really necessary or could any of the existing load models be used instead? So basically is the load model from Hisken so special that it should be included in the OpenIPSL?

lvanfretti, 2019-06-27:
@dietmarw I need to sit down and see if there is any difference in the load models, unless @ahsanKTH can tell us what is special about it, but probably it is just made to comply with the model used in the paper. Let me see if I get sometime this weekend.

Anyway, I think we can just call it "...Loads.Generic.LoadH" and put in the comments that this corresponds to the paper of Hiskens.

ahsanKTH, 2019-06-29:
@lvanfretti the load model is an exponential recovery load, with certain parameter settings. You are right I implemented same equations given in the paper of Hisken to get similar results.

lvanfretti, 2019-06-29:
@ahsanKTH thanks!
@dietmarw then I think we should name the load model: "...Loads.Generic.ExponentialRecovery"

marcelofcastro, 2019-07-31:
@ahsanKTH, @lvanfretti and @dietmarw, I checked the model for the exponential recovery model created by Ahsan and it is just another version of the already-implemented Loads.PSAT.ExponentialRecovery. 

Hiskens' paper used by Ahsan and PSAT manual use the same reference for exponential recovery loads: D. J. Hill, “Nonlinear dynamic load models with recovery for voltage stability studies,” IEEE Transactions on PowerSystems. 

The only difference between the model created by Ahsan and the existing model in OpenIPSL is that the state _xp_ is described differently. In addition, the implemented PSAT model has the option to include reactive power with exponential recovery. 

Therefore, I don't think is necessary to include Ahsan's load model into OpenIPSL.

lvanfretti, 2019-07-31:
@marcelofcastro ok with me.
Please do replace Ahsan's model with the PSAT model and parametrize it to get the same response as the one that Ahsan's example is giving.

marcelofcastro, 2019-08-05:
Ok @lvanfretti.
@dietmarw should I do this in your forked branch of OpenIPSL, where you already have Ahsan's system implemented?

dietmarw, 2019-08-05:
@marcelofcastro Today was my first day back at work. I will have a look during the week and let you know what I've already done and what still needs doing.

commit 516225b5 (2019-08-25): Merge pull request #192 from dietmarw/ULTC

Adding discrete ULTC (closes #189)
closed 2019-08-25 by marcelofcastro