OpenIPSL/OpenIPSL #194: new units improperly implemented in Threephase example

MaximeBaudette, 2019-09-11:
Hi, 

I was working with the Three phase package and got inconsistent results until I realized that proper units had been implemented in OpenIpsl, but it didn't make it to the example libraries. (@dietmarw I'm suspecting you did the push for the units ;-) ) 

Also I was wondering, shouldn't the network examples be moved back to the main library ? or at least in a different structure as Dymola won't load them except by doing it manually. 

I'll try to push a PR with some correction of the units in the Threephase package in the mean time.

dietmarw, 2019-09-11:
Yes correctly the examples have not yet been converted in the development version (i.e., master). There are several reasons for this:
 - It was planned to go through all examples to see which one should be kept, which ApplicationExamples should be included directly etc.
 - The examples also served the purpose of testing the automatic conversion for now. There are still some problems also on the tool side. 

I'm not sure what you mean with move **back** to the main library. Are you referring to ApplicationExamples? If yes then there is the same situation as for the OpenIPS.Examples for now.

If you like you can prepare a PR with the parameters adjusted to the now proper units.

MaximeBaudette, 2019-09-11:
Hi Dietmar, 

Yes I meant including the Application Examples directly into the main OpenIPSL package, at least for the standard network models. I know I made the argument before to keep them separated, but I feel like it was the wrong decision, as it is more difficult to keep them in sync. 

I'll prepare a PR with some of the updates I made manually on the three  phase package.

dietmarw, 2019-09-11:
> I know I made the argument before to keep them separated, but I feel like it was the wrong decision, as it is more difficult to keep them in sync.

You don't say :smile: 

> I'll prepare a PR with some of the updates I made manually on the three phase package.
:+1:

MaximeBaudette, 2019-11-13:
Alright, long overdue, also I made some error when pushing (pushed directly to the master of OpenIPSL) which I corrected.
You'll find my PR now.

lvanfretti, 2020-04-23:
@MaximeBaudette tic toc, almost end of April, will you get some time to fix this? Thanks!

marcelofcastro, 2020-05-21:
I'm taking over this issue

MaximeBaudette, 2020-05-21:
I did fix it, it's in the PR

marcelofcastro, 2020-05-21:
> I did fix it, it's in the PR

Yeah, I know. I am just trying to let @lvanfretti know that I am currently working on this issue, since I am working on the PR. Sorry if it was confusing...

closed 2020-06-09 by marcelofcastro