OpenIPSL/OpenIPSL #200: issues with new version of OpenModelica 1.14

syedyasserali, 2020-02-19:
there issues during simulations with new version of open modlica. they are
1) It is giving error like, components "not found in scope" or "not found in OpenIPSL". Many elements are not recognised which were used in the previous OpenModelica versions. Please refer to the attached figure 1.

2) The component values are changed or deleted in the latest OpenModelica version. Please refer to the attached figure 1.14.1
[image]
[image]

dietmarw, 2020-02-19:
It looks like you are using a newer version of OpenIPSL than the one you originally created. The latest stable version is: https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0

The model that you are getting an error on has been removed in the current development version and an improved version  `PwSunt` is to be used instead. When you work with the development version you will have to convert your library first. Unfortunately automatic conversion as provided by OpenIPSL is still not supported by OpenModelica. So you will have to update changed references by hand.

closed 2020-02-19 by dietmarw

lvanfretti, 2020-02-19:
@syedyasserali in other words, do not use the OpenIPSL library that ships with OpenModelica.
Download the stable version instead and use it: https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0

syedyasserali, 2020-02-20:
I have downloaded OpenIPSL from https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0 and after that i have tried 3 small networks, they are working perfectly. thanks for resolving issue.