OpenIPSL/OpenIPSL #201: Proper Commit Rules

lvanfretti, 2020-02-19:
@dietmarw @marcelofcastro @GiuseppeLaera 

Given recent conversations, mainly with PR #199 , I think we need to establish proper commit rules going forward to avoid confusion. 

The issue with trailing white spaces and dummy annotations seems to be already resolved by Dietmar in: [MSL #105 ](https://github.com/modelica/ModelicaStandardLibrary/issues/105).

So, I propose that before committing any code we ask the person making the PR to put it through the tool made by @dietmarw [Trimming of trailing white spaces](https://github.com/dietmarw/trimtrailingwhitespaces

@dietmarw do you have a Guideline, set of steps, that you can suggest other than this?

We don't need to add instructions to the main page now, that can be done with the edits made in the future in case anyone wants to make a commit other than the usual suspects...

dietmarw, 2020-02-19:
Ttws would not really have helped for #199. OpenIPSL is now in a one-file-per-class structure which makes it easy to only stage the files/classes that have been worked on. So the only guide line needed is to look and select what to stage for commit and not do a blind commit.

lvanfretti, 2020-02-19:
@dietmarw ok, but still, I would like you to write down "committing for dummies" instructions on how you want people to send this. The reason is that otherwise people will not fully understand where to start from, and it will create unnecessary new issues, maybe this can be simplified...

MaximeBaudette, 2020-04-26:
@dietmarw 
Can you add the TTWS to a PR CI to automatically run the script and commit the result onto pending PRs?

dietmarw, 2020-04-26:
It is not a good idea for the CI to make commits to the repository. Running bulk ttws would also do more harm to models that should not be touched since not part of the original PR. Also it looks like the current CI system does not work properly since it does not allow merging but does not report an problems either. Maybe at a first step it would be good to get the current CI system in shape again.

closed 2022-03-11 by marcelofcastro