OpenIPSL/OpenIPSL #205: PSAT Induction Machine Initialization

tinrabuzin, 2020-09-04:
Initialization of PSAT machine relies on the fact that machine's equations are included in the power flow calculations which result in the initial active and reactive powers (P_0 and Q_0). I think that as long as PSAT is used this is fine.

It might be good to include an option of compensating the reactive power with a shunt to match Q_0 which is not specified using the "correct" power flow equations. Otherwise the systems will not be properly initialized.

lvanfretti, 2020-09-04:
@tinrabuzin that's a good idea, I suggest that you can inherit the existing PSAT model and create a new one that includes this possible compensation. However, I don't know how the value of the compensation would be then provided by the user? Should we expect they do it with trial and error?
@marcelofcastro do you have any ideas on this? Manuel has been working with PSAT models, maybe he can help with this? Can you add him to this thread?

tinrabuzin, 2020-09-04:
@lvanfretti The compensation can be automatically determined during initialization. I'll try to include this soon. 

Btw, there is also a  startup flag in the induction machine models, but the functionality is not included there. We should maybe consider opening a new issue for that...

lvanfretti, 2020-09-04:
@tinrabuzin if it can be done through initialization, that would be ideal.

Yes, the start-up should probably be included, but we should be careful how to do that. Probably it is best to either set that as a "timed event" so that it does not create events that will speed down the simulation.

closed 2024-08-16 by lvanfretti