OpenIPSL/OpenIPSL #206: Changes of pfComponent's parameters during initialization

tinrabuzin, 2020-09-23:
I was trying to set the parameters of an element model that uses pfComponent in `initial equation` of the network model (basically via hand written power flow equations). 

But I was running into some issues when writing, e.g. `element.P_0 = P_powerflow`. The message that I get (in OpenModelica) is:
```
[12] 08:58:44 Translation Warning
The initial conditions are over specified. The following 1 initial equations are redundant, so they are removed from the initialization sytem:
         elemet.P_0 = P_powerflow
```

I added `P_0(fixed=false)` while extending pfComponent but the same message appeared. The problem was fixed only after I removed the default parameter value of P_0 in the pfComponent model itself. 

I'm not sure if the default values are needed in pfComponent, or what I am doing is a good way to do it, but we might want to consider enabling changes to the pfComponent's parameters.

dietmarw, 2020-09-23:
@tinrabuzin  `P_0` inside `pfComponent` is simply a parameter and has nothing with initialisation to do. So in order to understand your problem we'd need some more information on your model.

tinrabuzin, 2020-09-23:
@dietmarw I agree with you that it has nothing to do with initialization the way we've been modelling networks so far. But as I wrote above, I put a handwritten set of powerflow equations in the model that are solved during initialization and as a result output P_0 (among other variables). For instance, like this:
```
model network

  Element element1;
  Element element2;
  Line line;

  parameter Real P1=10e3;
  parameter Real Q1=10e3;
  parameter Real V1=1;
  parameter Real d1=0;
  parameter Real g = line.R/(line.R^2+line.X^2);
  parameter Real b = -line.X/(line.R^2+line.X^2);
  parameter Real P2(fixed=false);
  parameter Real Q2(fixed=false);
  parameter Real V2(fixed=false);
  parameter Real d2(fixed=false);
initial equation
// Powerflow equations
  P1 = g*V1^2 - V1*V2*(g*cos(d1-d2) + b*sin(d1-d2));
  Q1 = -b*V1^2 - V1*V2*(g*sin(d1-d2) - b*cos(d1-d2));
  -P2 = g*V2^2 - V1*V2*(g*cos(d2-d1) + b*sin(d2-d1));
  -Q2 = -b*V2^2 - V1*V2*(g*sin(d2-d1) - b*cos(d2-d1));
// Setting results of powerflow to elements
  element2.P_0 = P2;
  element2.Q_0 = Q2;
  element2.v_0 = V2;
  element2.angle_0 = d2;
equation
   connect(elment1.p, line.p )
   connect(elemet2.p, line.n)
end network;

```

dietmarw, 2020-09-23:
Still I have difficulties seeing where `pfComponents` comes into play here. It is not used in your code. So in order to debug one would have known what `Element` looks like since there you seem to have a overdetermined initialisation.

dietmarw, 2020-09-23:
Also `fixed=false` is not meaningful on a parameter. That only works on variables with start values.

tinrabuzin, 2020-09-23:
Elements are any kind of models in OpenIPSL that extend `pfComponents`. You can think of element1 being a generator like GENSAL and element2 being a ZIP load model. 

I don't see why the initialization is overdetermined. I have 8 initial equations and 8 unknown parameters. Thus why I have `fixed=false` with parameters. Also, I'm not sure `fixed` is restricted only to variables...

dietmarw, 2020-09-23:
Well sure you can use fixed=false on parameters but this just means that they might get random values assigned during initialisation which they then keep during the simulation. I don't think that is what you want though. 
Anyway, if you want to solve it I would need a minimal working example to diagnose.

tinrabuzin, 2020-09-23:
Yes, exactly for the reason to avoid random values is why I wrote the initial equation section.

dietmarw, 2020-09-23:
Yes as I said, fixed=false does not make sense when you at the same time provide binding equations. 
As for your overdetermined, please share a complete model where you get the error message.

closed 2021-02-07 by tinrabuzin

dietmarw, 2021-02-08:
@tinrabuzin  So is your issue resolved?