OpenIPSL/OpenIPSL #207: Error running N44 example

junjie-zh, 2020-10-07:
```
[6] 14:03:29 Translation Error
[N44.Original.Generators.Gen1_bus_7000: 5:3-27:77]: Variable G1_bus7000.gENROU: In modifier (V_0 = G1_bus7000.PSSE_data.voltages.V7000), class or component V_0 not found in <OpenIPSL.Electrical.Machines.PSSE.GENROU$G1_bus7000$gENROU>.

[7] 14:03:29 Translation Error
Error occurred while flattening model N44.Original.Nordic44_Original_Case

[8] 14:03:53 Translation Error
[N44.Original.Nordic44_Original_Case: 1365:5-1365:33]: Modified element V_0 not found in class Load.
```
Hi, with the newest commit I'm unable to run N44 example.  Looks like the `V_0` should be `v_0`? 
The similar error message is also reported to bus instances.
Is this a known issue?

dietmarw, 2020-10-07:
It looks like you are using the current development version instead of the last stable release. The examples have not yet been adapted to the newest development version. So I suggest you download the latest release: https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0 instead.

junjie-zh, 2020-10-07:
> It looks like you are using the current development version instead of the last stable release. The examples have not yet been adapted to the newest development version. So I suggest you download the latest release: https://github.com/OpenIPSL/OpenIPSL/releases/tag/v1.5.0 instead.

Many thanks for the quick reply! 
I checked out v1.5.0, the compilation succeeded, but the simulation (of `N44.Original.Nordic44_Original_Case`) is not converged, it looks like there's many zero divisions

```
solver will try to handle division by zero at time 0: trafo_5100_5101.t.im
division leads to inf or nan at time 0, (a=-14.6141) / (b=0), where divisor b is: trafo_5100_5101.t.im
solver will try to handle division by zero at time 0: trafo_6000_6001.t.im
division leads to inf or nan at time 0, (a=1874.64) / (b=0), where divisor b is: trafo_6000_6001.t.im
solver will try to handle division by zero at time 0: trafo_5500_5501.t.im
division leads to inf or nan at time 0, (a=28.1638) / (b=0), where divisor b is: trafo_5500_5501.t.im
...
```

dietmarw, 2020-10-08:
I tested it in Dymola 2020 and it works fine. So my guess it the problem is not necessarily on the OpenIPSL side but on the tool side. So you should contact the developers of the tool you were using and they might be able to help you further.

closed 2020-10-08 by dietmarw

junjie-zh, 2021-09-21:
Hi @dietmarw, I'm actually using OpenModelica to run this example and just noticed that you're one of the contributors :) Could you please try to run it with OpenModelica? In my case, OM 1.14 and 1.17.0 didn't work.

dietmarw, 2021-09-22:
@junjie-zh Like I said last time. Make sure you are running the example from the last official release which is still v1.5.0. I just tested it again and it runs (slowly) in OM 1.1.18 and Dymola2020.

junjie-zh, 2021-09-22:
Thanks again for the quick reply.
I've checked and pretty sure that the version is v1.5.0 (with `git log` I can see `commit 4b97ac8de5992289ad6327e3bd6dca4cdb985393 (HEAD, tag: v1.5.0)`)
The problem could also be from the simulation or translation flags for the OM, could you please share me your configurations? And perhaps the integration method, etc., in your simulation. That would be very helpful.
Also I'm working under Ubuntu 20.04 with kernel version 5.11.0-34-generic.

dietmarw, 2021-09-22:
But what is the Error you are experiencing? Mind it will take a **very** long time to simulate so it might look the model is stuck when really it is just slow.

junjie-zh, 2021-09-22:
At the beginning there were many zero-division warning like below:

> ```
> solver will try to handle division by zero at time 0: trafo_5100_5101.t.im
> division leads to inf or nan at time 0, (a=-14.6141) / (b=0), where divisor b is: trafo_5100_5101.t.im
> solver will try [...]

In the end the simulation fails with:

```
Solving non-linear system 8996 failed at time=0.
For more information please use -lv LOG_NLS.
Failed to solve the initialization problem with global homotopy with equidistant step size.
Unable to solve initialization problem.
simulation terminated by an assertion at initialization
```

For the solver I tried with implicit-euler, dassl, and ida. 

P.S. I switched to a Windows machine with OM-1.18.0, since its memory is larger than my ubuntu one. The version of openIPSL is still `v1.5.0`