OpenIPSL/OpenIPSL #217: Mark verified models via a distinctive icon

dietmarw, 2020-10-14:
I think it would be a good idea to mark models that are verified also visually. One idea which is the least intrusive would be to adopt what the MSL is doing for obsoleted models which simply extend from `Modelica.Icons.ObsoleteModel`. So for the verified models I would simply create an icon model `VerifiedModel` with a green stipulated frame which then all verified models can extend from.

lvanfretti, 2020-10-14:
Agree, something even prettier is what they do in the EPSL library, where they have a little rectangle with a color to designate the model fidelity (in this example they are all yellow though): https://github.com/modelica/newsletter/blob/master/_2018-01/EPSL.md
We could instead use a similar approach to designate if the model has been verified again in the work we are doing. 
Also, we should add to the model a link to where the verification results can be found, that's in another PR related to the documentation though...

dietmarw, 2020-10-14:
The problem with the rectangle is that it needs to be placed **on** the icon. Now this works if you have a unified system where there is unused space on all icons. This is not the case for OpenIPSL and would take quite some effort to do so. My first thought was actually a small green check mark in the corner of an icon. So having the green frame would at least work on all icons.

lvanfretti, 2020-10-14:
Oh, then we go for the easiest solution.
I think I need to get some graphic designer to re-do all icons for us in
the future.

-- 
Luigi Vanfretti, PhD, Associate Professor
ALSETLab.com <http://alsetlab.com/> - ECSE <https://www.ecse.rpi.edu/>, RPI
<http://rpi.edu/>, Troy, NY, USA <https://goo.gl/maps/mVowWWPLtL72>
E-mail: luigi.vanfretti@gmail.com
iPhone: +1 (518) 496 0196 - Skype: lvanfretti
Github <https://github.com/lvanfretti>  -  Google Scholar
<https://scholar.google.com/citations?user=_XI6Q6UAAAAJ&hl=en&oi=ao>  -
ORCID <https://orcid.org/0000-0002-4125-1055>  -  LinkedIn
<https://www.linkedin.com/in/vanfretti/>  -  Twitter
<https://twitter.com/lvanfretti>  -  Youtube
<https://www.youtube.com/user/lvanfretti>

dietmarw, 2020-10-14:
I think you can use your energy on better things than fighting with designers ;-)

dietmarw, 2020-10-15:
I started working on this. What a can of worms I've opened with that .... :-(

marcelofcastro, 2020-10-15:
@dietmarw what happened?

lvanfretti, 2020-10-16:
@dietmarw I hope it is not too much of a problem, sorry to hear this.

PR #228 (2020-10-17): Mark verified models via a distinctive icon
This fixes #217.

This at the same time also cleans up most of the models that are using the verified models.

The only one not yet done is GENCLS because of #227
closed 2020-10-17 by dietmarw