OpenIPSL/OpenIPSL #220: CI setup needs to be updated

dietmarw, 2020-10-14:
The current set of docker images and CI test scripts fail. Without investigating in detail just looking at the docker files and scripts the following issues should be fixed: 
- `Dockerfile` 
  - should be put under `CI/`
  - is loading MSL 3.2.2 when OpenIPSL is at 3.2.3
  - updating to ubuntu 18.04 (i.e., `phusion/baseimage:1.0.0`) would be a good idea
- python scripts
  - python 2 is discontinued so use this chance to finally update the scripts to python 3

dietmarw, 2020-10-14:
@MaximeBaudette  could you give me access rights to Travis so I can try to fix some of the setup?

marcelofcastro, 2020-10-16:
@MaximeBaudette and @dietmarw do you guys have any idea why the syntax check/ travis build stopped working?

dietmarw, 2020-10-20:
So I fixed it all in #224. Ideally we could use GitHub Actions but the syntax of the setup is quite a bit different to travis.  @mannuelnvro If interested you can have a look at #234 at my pathetic attempts to get that going. But maybe even better to start from scratch.

dietmarw, 2020-10-20:
I actually forgot to move the Dockerfile but probably leave that for now.

PR #224 (2020-10-20): Fix the CI setup
I'm using the PR in order to fix the CI setup again.
Fixes #220
closed 2020-10-20 by dietmarw