OpenIPSL/OpenIPSL #226: GGOV1 needs to get cleaned up.

dietmarw, 2020-10-16:
`Electrical.Controls.PSSE.TG.GGOV1.GGOV1` needs to be moved one level up and the support models for that block put under `BaseClasses`.  At the same time the documentation needs to get fixed for the support models.

dietmarw, 2020-10-17:
Also good to look at #119 at the same time.

lvanfretti, 2020-10-27:
@dietmarw other than the issue in #119 on the connection, another thing that Miguel did was to look at the diagram and try to reorganize the whole mess into logical pieces, i.e. what each part of the diagram represents, and then separate it in blocks. The final version of that model is in this repo: [link](https://github.com/ALSETLab/2018_AmericanModelicaConf_PowerGrid_plus_PowerSystems).

lvanfretti, 2020-10-27:
@marcelofcastro this would be a good assignment for Fernando so he learns the workflow of cleaning up things with you. Can you discuss with him and have him help @dietmarw ?

marcelofcastro, 2020-11-25:
Issue solved with PR #254

closed 2020-11-25 by marcelofcastro