OpenIPSL/OpenIPSL #227: Possible unit issue in GENCLS

dietmarw, 2020-10-16:
Whilst cleaning up I found this in GENCLS:
```mo
Real delta(start=delta0, fixed=true) "Rotor angle (deg)";
```
whilst later `delta0` is calculated:
```mo
parameter SI.Angle delta0=atan2(vi0 + R_a*ii0 + X_d*ir0, vr0 + R_a*ir0 - X_d*ii0);
```
and is clearly rad. I think the type of `delta` also needs to be `SI.Angle` and the "(deg)" comment needs to be removed. If the user should enter the value in deg then the displayUnit annotation should be used instead.

dietmarw, 2020-10-16:
I'm holding of the "verified" certification for that class until this issue has been resolved.

marcelofcastro, 2020-10-16:
you're right!

dietmarw, 2020-10-16:
OK in that case I'll fix this inside #228.

closed 2020-10-17 by dietmarw