OpenIPSL/OpenIPSL #245: OM Compatibility

marcelofcastro, 2020-11-11:
Models such as PSS2A, PSS2B, WT4E, EXST1 and STABNI are not working in OpenModelica. Need to fix it for compatibility with open-source tool.

marcelofcastro, 2020-12-15:
ESAC2A should also be investigated.

marcelofcastro, 2022-05-24:
GGOV1 and GGOVDU1 should have their implementation enhanced (probably with the better implementation of basic blocks), as discussed in PR #290

dietmarw, 2024-11-25:
@marcelofcastro Any chance to have another look or decided if this is actually still relevant for the 3.1.0 release (or at all).

marcelofcastro, 2024-11-29:
Yes! I want to take a look at it again since we had some re-implementation of basic blocks. It might not be relevant to block the release, but it is definitely worth checking it.

dietmarw, 2026-01-15:
So should this issue be bumped to milestone:3.2.0?

dietmarw, 2026-10-23:
@lvanfretti Wanna point Claude at this issue?

lvanfretti, 2026-10-23:
@dietmarw yes, these are PSS/E models, the prototype for that is in the works, and I'll prioritize this as I tune it.