OpenIPSL/OpenIPSL #247: Version 2.0.0 Release

marcelofcastro, 2020-11-11:
### Issues to solve for 2.0.0 release:

- [ ] #246 
- [ ] #245 
- [ ] #226 
- [ ] #179 
- [ ] #165 
- [ ] #162 
- [ ] #160 
- [ ] #119 
- [ ] #67

lvanfretti, 2020-11-19:
@dietmarw @marcelofcastro can we add: 
- modify/update the readme.md. We need to synthesize and move materials that are not needed in the landing markdown to others. I think it would be great if it could look like other Github repositories, and move other things accordingly.
- add funding information. @marcelofcastro we need to add the support of NYPA/NYSERDA and CURENT, I need to add OpenCPS support during a year or so, and @dietmarw should add anything that is supporting him to contribute to this effort.

dietmarw, 2020-11-19:
> and @dietmarw should add anything that is supporting him to contribute to this effort.

OK I'll try to include a personal paypal donate button then. :-D

PR #297 (2022-06-08): Adding the release notes
closes #247 

@marcelofcastro and @lvanfretti : Have a look if you think that I got everything caught in the release notes. I don't think we need to update the readme itself but rather extend the release notes on the GitHub Releases page once this is accepted.
closed 2022-06-08 by dietmarw