OpenIPSL/OpenIPSL #248: DisplayPF: voltage and angle values are not display in buses

JosifFP, 2020-11-16:
Hi!,
I am starting in OpenIPSL and OMEdit, learning from examples and tutorials.
I have not resolved and understood why PF values in buses are not displayed in my examples (IEEE9 Bus)
[IEEE9Bus.zip](https://github.com/OpenIPSL/OpenIPSL/files/5544533/IEEE9Bus.zip)

I was using OpenModelica 1.16.0 and OpenIPSL 1.5.0. for what I have mentioned before.
I tried with OpenModelica 1.13.0 and OpenIPSL 1.5.0, and at least I could see "Vpu, Angle".
In the plotting window, steady-state values (V and angle) are shown, so the problem is not the simulation itself.
Am I missing something? or values are not usually displayed OMEdit.
_______________________________________________________________________________________________________________________

dietmarw, 2020-11-16:
Unfortunately the support of [Variable Graphics and Schematic Animation](https://specification.modelica.org/v3.4/Ch18.html#variable-graphics-and-schematic-animation) is currently extremely limited in OMEdit. You can follow the progress in [Open Modelica #5631](https://trac.openmodelica.org/OpenModelica/ticket/5631). So these features will not work in OMEdit. They do work as expected in Dymola (have not tested in other tools).

closed 2020-11-16 by dietmarw

JosifFP, 2020-11-16:
Thank you for your answer, Dietmar.