OpenIPSL/OpenIPSL #250: Introduce normalisation in addition to SI units?

dietmarw, 2020-11-19:
When looking at the [PowerGrids](https://github.com/PowerGrids/PowerGrids) library (which basically is a different approach to reimplementing iPSL but making the most of Modelica functionality) I noticed that they are not only using SI and pu but make also use of the `nominal` attribute. What this basically does is telling the solver in which range numbers will "live" and hence avoiding numerical issues when trying to handle large numbers (e.g., power in the 1e8 range or voltage in the 1e5 range) and per unit (0 to 1) at the same time.  So I got talking to @casella and asked him of the real benefit and he pointed me to a paper he had written on this: https://doi.org/10.1145/3158191.3158192

Making the extra step to introduce nominal values is actually not such a big deal (now that we already have most of the values converted to SI). We could simply use the same approach that PowerGrids used by defining our own types (see https://github.com/PowerGrids/PowerGrids/blob/f51f4c4f0d13f3167b6670df2fb17520feaf3170/PowerGrids/Types.mo#L5-L20 ). The update is a trivial find replace. 
Any objections to do this extension for 2.0.0?

tinrabuzin, 2020-11-19:
Just to add to this, I noticed significant improvements during initialization of my models (which partially use OpenIPSL) when I set nominal values.

lvanfretti, 2020-11-19:
@dietmarw I don't know if it's a good idea for version 2.0.0, maybe we can put that latter... it depends on how much time you can put into it, we definitely want to release the new version of the library before the end of the year. What do you think @marcelofcastro ?

dietmarw, 2020-11-19:
That change takes (me) about 15 mins to incorporate.

lvanfretti, 2020-11-19:
@dietmarw then if that's the case, that would be great to have! 
We need somehow to compile a list of main improvements, so let's not forget this one!

dietmarw, 2020-11-19:
> We need somehow to compile a list of main improvements, so let's not forget this one!
That's what we have PR and issues for here ;-)

lvanfretti, 2020-11-19:
@dietmarw awesome!

dietmarw, 2020-11-19:
I think I managed it also in time :-)

lvanfretti, 2020-11-19:
Well, I am very happy that we have your help in this library @dietmarw ! Thank you for everything you are doing!

casella, 2020-11-19:
> 
> 
> Just to add to this, I noticed significant improvements during initialization of my models (which partially use OpenIPSL) when I set nominal values.

Given our analysis, I'm hardly surprised 😄

PR #252 (2020-11-20): Introduce nominal values to help numerical solvers
Fixes #250.
closed 2020-11-20 by dietmarw