OpenIPSL/OpenIPSL #256: GENCLS model needs to get investigated

dietmarw, 2020-11-26:
There is something not quite right with the `GENCLS` model. It is supposed to be a constant voltage source but contains parameters like resistance and reactance which is conflicting information. So either the model or the documentation needs to be updated.

marcelofcastro, 2020-11-26:
No it is not a constant voltage source. It is classical generator model. Which means classical swing equations. It works as a constant voltage source (infinite bus) if you set inertia to zero. Documentation might be just describing the infinite bus role, but model is correct.

dietmarw, 2020-11-26:
Not so fast with closing this one. Please have a look at the model and where it is used in examples first. It currently has protected parameters which lead to double initialisation. This lead me to opening this ticket since there are parameters used/defined which should not have to be defined if the model does what the documentation says.

marcelofcastro, 2020-11-28:
Are you sure that GENCLS is the problem? Documentation is actually correct: 

> Synchronous machine represented by 'classical' modeling or Thevenin Voltage Source to play Back known voltage/frequency signal.

Resistance and Reactance are used in both so nothing wrong regarding parameters.

When I displayed conflicting initial variables were actually pointing to currents in the lines that have basically two options:
- Machine precision from start values in `Interfaces.PwPin` (something that we did in the past, don't remember why)
- Nonlinear solver PF solution calculated from voltage magnitudes and angles in machines, loads and (luckily) buses.

The conflicting variables disappear if we remove the machine precision forced initialization.

closed 2020-12-01 by dietmarw