OpenIPSL/OpenIPSL #260: PSSE WT1G models accidentally removed.

dietmarw, 2020-12-10:
Currently the `WT1G` models are no longer present. These have been removed as part of #151 in order to fix #148. But it looks like the WT1G models should never have been part of that PR. Anyway, in case those models still have value in addition to the `WT3G` and `WT4G` models that are currently present then the links given above should help recovering those.

dietmarw, 2023-01-13:
@marcelofcastro and @lvanfretti Should `WT1G` get resurrected?

lvanfretti, 2023-01-17:
@dietmarw I would like to dig in to see what they are and if they can be validated against PSS/E, but we need a volunteer for that. I think we can discuss further offline.

dietmarw, 2024-11-16:
> @marcelofcastro and @lvanfretti Should `WT1G` get resurrected?

@lvanfretti , @marcelofcastro ?

lvanfretti, 2024-11-17:
@dietmarw wind turbines based on WT1G models are really old and may not even be in operation any longer.
Other than for academic or "historical" value, I'm not sure it is useful to keep them. From what I see, most turbines are now type 3 (based on WT3G) and newer ones are WT4G (but the cost is high). Also, the WECC models that we have #327 are going to be the "industry standard" going forward.

So, in summary, only if you think we should have it for teaching or digital archeology purposes, it is worth to resurrect this.

dietmarw, 2024-11-17:
Probably not then.

closed 2024-11-17 by dietmarw