OpenIPSL/OpenIPSL #263: OCT Compliance

lvanfretti, 2020-12-17:
We are working with Modelon to make OpenIPSL compliant with the Optimica Compiler Toolkit, used in several tools, e.g. Modelon Impact, ANSYS Twin Builder, etc.

Below is the summary of issues from Hubertus Tummscheit when we did some live testing on 12/16/2020:
> Here is what I changed, and I updated to this evenings version on github and made sure these are minimal edits to make things run:
> Loads.PSSE.BaseClassse.baseload.mo
> Machines.PSSE.GENROE.mo
> All changes in the above are to create Complex parameters as Complex(a,b) and not as a + j*b
> With the above PSS2A runs.
>
> Branches.Generic.ULTC (added 2 Boolean variables, and added a pre in when-clause)
> It would be easier if you had true per-component unit tests, which you may have somewhere.

Fixes proposed:

- [ ] Fix: Loads.PSSE.BaseClassse.baseload.mo, a per attachment in zip.
- [ ] Fix: Machines.PSSE.GENROE.mo, a per attachment in zip.
- [ ] Fix: ULTC.mo, a per attachment in zip.

I'm assigning Giuseppe to work with this after we install and test the OCT on our systems.

[OCT_Tests.zip](https://github.com/OpenIPSL/OpenIPSL/files/5706324/OCT_Tests.zip)

dietmarw, 2020-12-17:
That looks sensible. 

> I'm assigning Giuseppe to work with this after we install and test the OCT on our systems.

Well then you should do so ;-) 
[image]

dietmarw, 2020-12-17:
The changes could still be included in version 2.0.0.

lvanfretti, 2020-12-17:
Thanks, Giuseppe was not in the team yet, so I had to add him and wait for the invitation. 
Thanks for doing it!

lvanfretti, 2020-12-17:
> The changes could still be included in version 2.0.0.

These ones yes, but the ones that will come up from them testing with OCT, is better we bulk them into a single release.

lvanfretti, 2020-12-17:
> The changes could still be included in version 2.0.0.

@dietmarw yes, I think so, but only for these three. We just got access to the latest release of OCT, but we need to install it and do the testing, etc.

The whole testing on the library for OCT compliance will be carried out next semester by @ManuelNvro and @GiuseppeLaera .

However, the debugging will have to wait until May/June when/if @GiuseppeLaera does his internship in Modelon. He is working out this with Hubertus. We are hoping this takes place so that we can get help from Modelon's team when we get strange errors from their compiler that we are not familiar with, and from Hubertus since some of the errors detected here really needed someone that knows well the language spec.

hubertus65, 2020-12-29:
BTW: I found the reason why the initialization in Dymola and OCT gives different results. In the model "OpenIPSL.Examples.Machines.PSSE.GENROE", The components GEN1.p.vi and gENROE.p.vi (and vr respectively) are the same variables, and initialization can be chosen once in gENROE with an angle_0 of 4.038907 degrees, and in GEN1 with and angle_0 of 0.0 degrees. Both are possible and legal, OCT chose 1, and Dymola the other. Setting both the same (to whatever it should be) should give the same result. This does not solve the issue that this problem is easy to create for users, but probably the cases of different results. Dymola flagged this as redundant initialization, btw., when translating in pedantic mode. FWIW: my changes to make the example run in OCT at all _should_ only have been syntactic changes, no changes to semantics. But of course, there could be an error by me.

hubertus65, 2020-12-30:
So: there was an error by me, I made a sign error in changing the syntx. In the linked pull request, both the issue with ULTC, and the Syntax problems in OCT are both fixed. See: https://github.com/OpenIPSL/OpenIPSL/pull/265. But please take a look at the changes to the ULTC model, which are documented in the committ message in detail.

lvanfretti, 2021-01-04:
@GiuseppeLaera can you please follow up with this and #265 ?

hubertus65, 2021-01-09:
@GiuseppeLaera: I looked some more at some errors, and found a class of errors (illegal comparison between Reals), for which I have a good idea on a better design pattern for replacing it. Holler once you start working on this.

hubertus65, 2022-06-07:
Hi @marcelofcastro: why did you move this to release 3.1.0 again?

dietmarw, 2022-06-08:
@hubertus65 It was decided to not include this in the 2.0.0 release in order that 2.0.0 can be released soon (this will actually happen today). At the same time, we decided to release version 3.0.0 immediately afterwards. The difference is that 3.0.0 will use MSL 4.0.0. Any future extension will then happen on the MSL 4.0.0 based versions, i.e., 3.x.0. I hope that clarifies this. 
Once 3.0.0 is released I will rebase the remaining PRs that are (now) scheduled for 3.1.0 onto the new master.

hubertus65, 2022-06-08:
I see and thought something like that was up. It makes sense, we are on MSL 4.0.0 anyways. I hope that all the changes that Giuseppe did are easy to upgrade to MSL .0. They should, ...

lvanfretti, 2022-06-08:
@hubertus65 one thing that has been holding @GiuseppeLaera up is access to the computer he used at Modelon, so he can get all the models out of it and so on. If you can help with that, then it will be more likely all improvements can be included.

closed 2023-01-12 by dietmarw