OpenIPSL/OpenIPSL #264: Changing License to BSD-3-Clause

lvanfretti, 2020-12-17:
To align to the Modelica Standard Library, it is necessary to figure out how to change the license to BSD-3-Clause https://opensource.org/licenses/BSD-3-Clause

I will seek for permission from iPSL license holders to do this change. 
@dietmarw I presume an email confirmation that we store with this issue can be sufficient?

Let me know what we need to do to achieve this if you have any insight.

@GiuseppeLaera please help me monitor this issue and keep up the follow up for this.

dietmarw, 2020-12-17:
There is no need. MPL 2.0 is perfectly compatible with BSD

lvanfretti, 2020-12-17:
@dietmarw there are two reasons from what I understand, that would require this change:
-  Not having the same license would put some difficulties into "repackaging" the software together with the MSL. This is needed for the research project I have with DOE where we need to release some "self-contained" models.
- The change of the MSL seems to have been driven by the needs of the Buildings library to meet Department of Energy and University of Berkley requirements. So, for me to try to get some funding from DOE will lead to the same requirement, and I want to start dealing with this sooner than latter.

So, what do you think given this feedback? @dietmarw

hubertus65, 2020-12-17:
I can add a bit of flavor to this: in past discussions, DOE/DOE funded lab lawyers have been very strict about not allowing any "copyleft" licenses in the work they fund. For other projects that I am involved in, MSL not being BSD (or, more loosely, anything non-copyleft + open-source) was a hard requirement.

dietmarw, 2020-12-17:
> For other projects that I am involved in, MSL not being BSD (or, more loosely, anything non-copyleft + open-source) was a hard requirement.

Could you clarify that. MSL **not being** BSD was a hard requirement? So for those project it was a bad thing that MSL is BSD?

hubertus65, 2020-12-17:
Continued funding for Michael Wetter's projects around the next generation building simulation tool by DOE, tool codename SOEP or Spawn of Energy Plus, was conditional on MSL changing its license or at least dual-license. Otherwise, DOE would and could according to their rules not package MSL + Buildings library in one package for redistribution. I edited the original code for better clarity.

lvanfretti, 2021-12-04:
Approvals from RTE, AIA and DTU were obtained as per the attached .pdf
[License Agreement for OpenIPSL.pdf](https://github.com/OpenIPSL/OpenIPSL/files/7654223/License.Agreement.for.OpenIPSL.pdf).

Following the dev meeting with @dietmarw and @marcelofcastro, the license will be changed to BSD3 as per #281.

closed 2022-03-11 by marcelofcastro