OpenIPSL/OpenIPSL #269: ApplicationExample IEEE_9_Buses misses VoltDependant Load

sorrento, 2021-03-04:
In line 28 of 
OpenIPSL/OpenIPSL/ApplicationExamples/IEEE9/IEEE_9_Buses.mo

` OpenIPSL.Electrical.Loads.PSAT.VoltDependant`

I understand it shoud say:
`OpenIPSL.Electrical.Loads.PSAT.VoltageDependent`

which is a model inside the library

dietmarw, 2021-03-04:
Yes this was renamed (and corrected the spelling) for 2.0.0 dev but nut included in the conversion rules it seems. So might be a good idea to add it there.

sorrento, 2021-03-04:
It happens the same in IEEE_14.  It seems it also changed the parameter V_0 ->v_0

dietmarw, 2021-03-04:
Yes there are a lot of changes since 1.5.0 hence the application examples need to get carefully investigated in order to move them into OpenIPSL

pipeacosta, 2021-05-05:
Hi. I will post it here because I think it could be related to this issue. In the TwoAreas application example:
- PSAT: Warning: maximal number of iteration reached but no root found. nonlinear system 429 fails: at t=0
- PSSE: The same warning, followed by: "The initialization finished successfully without homotopy method.", after that:
division by zero in partial equation: sqrt(Line9_10.vs.re ^ 2.0 + Line9_10.vs.im ^ 2.0)
at Time=1.520000
solver will try to handle that.
Results show that it seems to be a numerical instability. 

However, I don't know how to relate these issues with changes in the library, could you give me an advice?

Thank you in advance!

dietmarw, 2021-05-06:
Hi @pipeacosta 
My advice would be to stick with the last official released OpenIPSL 1.5.0 until the final 2.0.0 is released later this year. That will then also contain the updated application examples (which also will mainly focus on the PSSE variants). Now in OpenIPSL 1.5.0 the PSAT model simulates w/o problems, both in OpenModelica and Dymola. The PSSE version however will only succeed in Dymola and fail in OpenModelica. So here it is unfortunately a tool issue.

marcelofcastro, 2022-05-06:
@dietmarw I believe this issue is solved with PR #286

dietmarw, 2022-05-09:
Yes, I agree.